realsenseai/librealsense #8305: Installing on a Raspberry Pi 4b running Ubuntu Desktop 20.04 libUVC

matthewjreeves, 2021-02-03:
I have tried numerous ways to install the Realsense SDK on a RPI 4B (literally allsorts, but I am not a Linux whizz, everything I do has to be done with the help of online resources and pretty much every method I found failed in some way or another)
I finally found the libUVC method which worked perfectly with Raspbian Buster, however my end game is to run in on a 4B running Ubuntu 20.04.

I noticed at the bottom of the livUVC page, it asks for suggestions for systems it could be expanded to support. 

Would it be possible to support Ubuntu 20.04. It seems to be hitting a problem with the compiler I have had on numerous other attempts to install. THe error is below.

Many thanks in advance for any help you can give.

+ cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release
-- The CXX compiler identification is unknown
-- The C compiler identification is GNU 10.2.0
CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring incomplete, errors occurred!

Error log below

[CMakeError.log](https://github.com/IntelRealSense/librealsense/files/5920797/CMakeError.log)

MartyG-RealSense, 2021-02-03:
H @matthewjreeves   Which librealsense SDK version are you currently using, please?  Official support for Ubuntu 20 was added in SDK version **2.40.0**.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

matthewjreeves, 2021-02-03:
Hi @MartyG-RealSense I have just retraced my steps (Had been trying to follow https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049)
and it had directed me to V 2.41.0. I think the problem with this method was that I couldn't get cmake to update to work with the compiler in Ubuntu 20.04

MartyG-RealSense, 2021-02-03:
https://github.com/IntelRealSense/librealsense/issues/6964 would be more suited for Nvidia Jetson boards.  The **Acrobotic** insstallation guide for performing a **-DFORCE_LIBUVC=true** build with CMake on Pi 4 may work better for you. 

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

matthewjreeves, 2021-02-03:
I tried that method. It seems to be aimed at RPI OS. It failed at the autogen stage in installing protobuf on Ubuntu. I couldn't find a solution to that one. Maybe I need to persevere with that method?

matthewjreeves, 2021-02-03:
I will give that a go tomorrow. So much hair tearing!
Is there any chance the libuvc backend (https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md) will be updated so this whole process is a little easier for people using the most common SBC out there with Ubuntu?

MartyG-RealSense, 2021-02-03:
I am not aware of plans to update the **libuvc_installation** guide.

The basic history of the situation is that up until SDK 2.30.0,  **-DFORCE_LIBUVC=true** was used.  From that SDK onward, the **RSUSB** install method was introduced. This uses the CMake term **-DFORCE_RSUSB_BACKEND=true**

The CMake build flag list at the link below illustrates that DFORCE_LIBUVC is still a valid build flag **but** it is recommended to use DFORCE_RSUSB_BACKEND instead.

https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration

Typically, the main place that I still see DFORCE_LIBUVC in use is on Raspberry Pi builds, whilst DFORCE_RSUSB_BACKEND is commonly used for PC, Jetson, etc.

matthewjreeves, 2021-02-09:
MartyG

Many thanks, your pointers certainly got me going in the right direction. There were a lot more failed attempts an a lot of Googling, but I have managed to get the Realsense SDK and ROS node up nad running. Often have to unplug and replug my T265 to get it to recognise it, but apart from that, perfect.

Many thanks for your assistance!

closed 2021-02-09 by matthewjreeves

MartyG-RealSense, 2021-02-09:
You are very welcome @matthewjreeves - it is excellent news that you developed a solution.   Thanks for the update!