realsenseai/librealsense #8306: Is there any way to remove depth data of target's side?

hailsong, 2021-02-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I'm developing example for reconstruct 3d pointcloud of target and I used ICP (point to plane). I worked well but I found some pointcloud of target's side make the output bad. It's maybe the background noise of realsense and I tried to apply removing outlier tool but I didn't work.
Here's photo of my output
[image]

My question is 'Is there any tool to remove pointcloud of target's side?' In 2D image, I used morphology operation for erode... is there any tool like this?

MartyG-RealSense, 2021-02-04:
Hi @hailsong   I considered your iage carefully  My understading of statistical outlier filtering is that it removes detached floating clumps of point cloud data that are separated from the main image.  Therefore it may not help if the points on an edge are untidy but firmly connected to the main body of the cloud.

Could you describe the method that you are usin to generate the Open3D cloud with RealSense please?

hailsong, 2021-02-09:
@MartyG-RealSense Hi! Thanks for responding.
Firstly, I got pointclouds through pipeline and cropped my ROI (in this case, the target was a person). I saved it in .ply file.
[image]
(pointcloud from realsense)
[image]
(cropped pointcloud data, I saved it to .ply file)
Than I opened .ply file through open3d.io.read_point_cloud and apply ICP Multiway registration.
And I found some noise of pointcloud and I guessed this is noise from background (or maybe motion blur?)
[image]
gray pointclouds are the trash pointcloud (it's on my nose) (and its color is from background. my labs wall is gray color.)

hailsong, 2021-02-09:
@MartyG-RealSense
I tried realsense's built-in filter RSM - threshold remove (?) to reduce this kind of noise.
[image]

I can reduce it but I lose some correct pointcloud data..

MartyG-RealSense, 2021-02-09:
If you are creating the ply file in the RealSense Viewer and then opening it in Open3D, you could try setting the **Depth Units** option in the Viewer to **0.0001** instead of the default 0.001.  This will create a more zoomed-in view but can reduce depth noise significantly.

You can find the Depth Units option under the **Controls** sub-section of the **Stereo Module** options

In the case in the link below, I used a 0.0001 depth unit scale to create a textured point cloud in the Viewer's 3D mode that had much less noise than when using the default 0.001 scale.

https://github.com/IntelRealSense/librealsense/issues/8228

hailsong, 2021-02-10:
[image]
Your solution makes it better! but I still have some noise of it.

MartyG-RealSense, 2021-02-10:
Under the **Depth Controls** section of the **Advanced Controls** options in the Viewer, does setting **DS Second Peak Threshold** to zero reduce the amount of loose dots on your image?

[image]

MartyG-RealSense, 2021-02-16:
Hi @hailsong  Do you require further assistance with this case, please?   Thanks!

hailsong, 2021-02-18:
@MartyG-RealSense thanks a lot!

MartyG-RealSense, 2021-02-18:
Is your problem solved now, please?  Thanks!

closed 2021-03-01 by MartyG-RealSense