realsenseai/librealsense #8309: pointcloud looks not such smooth

coldbath, 2021-02-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Operating System & Version |   Linux Ubuntu 16.04  |                                        | 
| Platform                                 | PC |

### Issue Description
hi,
I have calibrated the camera with On-Chip Calibration and Tare Calibration, and set disparity shift to 0.001.
When I capture depth and rgb, found that the depth values unstable, the wall pointcloud looks not such smooth.
[image]
[image]

But when capture the depth image with matterport depth camera, the wall pointcloud looks very smooth and dense.

[image]
[image]

I wonder the phenomenon caused by camera precision limit or other reasons, does any one can give some suggestion？

thx.

MartyG-RealSense, 2021-02-04:
Hi @coldbath  I have seen a similar case recently in regard to the step-shaped gap in the depth image.  That was on a D410 camera built into a tablet that was running Windows, which would suggest that this issue is not related to the D455 camera hardware or the operating system.  The step only occurred when aligning depth to color, and was not present in non-aligned images.

https://github.com/IntelRealSense/librealsense/issues/8216

Does the image noticably improve if you use a smaller depth unit scale of **0.0001** please?

coldbath, 2021-02-05:
> Hi @coldbath I have seen a similar case recently in regard to the step-shaped gap in the depth image, which was also present on the infrared image. That was on a D410 camera built into a tablet that was running Windows, which would suggest that this issue is not related to the D455 camera hardware or the operating system. The step only occurred when aligning depth to color, and was not present when alignment was removed.
> 
> #8216
> 
> [image]
> 
> Is the step-shaped gap also present on the infrared image, please?
> 
> Also, does the image noticably improve if you use a smaller depth unit scale of **0.0001** please?

Thx for your reply, 

Firstly, the image noticably improve when use a smaller depth unit scale of 0.0001.

And, the infrared image still exist step-shaped gap.

[image]
[image]

MartyG-RealSense, 2021-02-05:
I will seek Intel's advice about the issue of the step-shaped gap in the infrared frame.  Thanks very much for your patience.

MartyG-RealSense, 2021-02-06:
Hi again @coldbath  The advice that I receivd from Intel was in agreement that the problem was related to the alignment process.  This taalies with the experience of the RealSense user in https://github.com/IntelRealSense/librealsense/issues/8216 who found that the step shape disappeared from the image when alignment was removed.

coldbath, 2021-02-10:
> Hi again @coldbath The advice that I received from Intel was in agreement that the problem with the 'step' shape was related to the alignment process and that the step was caused by the depth map. This tallies with the experience of the RealSense user in #8216 who found that the step shape disappeared from the image when alignment was removed.
Hi @MartyG-RealSense When I capture the data, just open the Stereo Module (RGB Camera is closed), does this situation still opened the alignment? And I wonder how to remove the alignment, does the Intel RealSense Viewer has a button, or need to set with SDK?

coldbath, 2021-02-10:
> Hi again @coldbath The advice that I received from Intel was in agreement that the problem with the 'step' shape was related to the alignment process and that the step was caused by the depth map. [...]

Hi @MartyG-RealSense When I capture the data, just open the Stereo Module (RGB Camera is closed), does this situation still opened the alignment? And I wonder how to remove the alignment, does the Intel RealSense Viewer has a button, or need to set with SDK?

MartyG-RealSense, 2021-02-10:
If you are only activating the Stereo Module depth stream in the Viewer's 3D point cloud mode then there is no mapping between depth and RGB.

[image]

When RGB is activated in the RGB Camera module whilst in 3D mode, RGB maps to the point cloud.  Disabling the RGB stream removes that mapping.

[image]

If you have both depth and RGB enabled and only want the depth in the 3D point cloud mode, you can use the **Texture** option and select **Depth** from the drop-down menu.

[image]

kaosmaja, 2021-02-11:
Hi @MartyG-RealSense, I have a problem related to smooth surfaces appearing wavy (like the wall in your picture), is there a way to make the computer to see this as smooth?

MartyG-RealSense, 2021-02-11:
Hi @kaosmaja  Could you confirm which of the above images you are looking at please?   Thanks!

kaosmaja, 2021-02-11:
Hi @MartyG-RealSense, I'm referring to the depth image. Is there a way to get rid of the waviness on the left? (given a frame or video)

MartyG-RealSense, 2021-02-11:
A rough left hand side like on this image?

[image]

kaosmaja, 2021-02-11:
No, this is a couple of meters ahead in the front.

MartyG-RealSense, 2021-02-12:
@kaosmaja   I am sorry, I cannot identify the type of problem that you have without an image so that I can see what you are seeing.

Do the information resources in the link below provide useful insights about how to improve your depth image?

https://support.intelrealsense.com/hc/en-us/community/posts/360051121653/comments/360014076833

MartyG-RealSense, 2021-02-19:
Hi @coldbath and @kaosmaja   Do either of you require further assistance with this case, please?  Thanks!

closed 2021-02-25 by MartyG-RealSense