realsenseai/librealsense #8310: RealSense Viewer with lots of frame drops

Borzborz, 2021-02-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.03.00 | 
| Operating System & Version |   Ubuntu MATE 20.10 aarch64   | 
| Kernel Version (Linux Only)    | 5.8.0-1013-raspi  | 
| Platform                                 | Raspberry Pi 4B 8GB  |
| SDK Version                            | 2.41.0   | 
| Language                            | Python (3.8.6) | 
 

### Issue Description

I have been searching for the issues but can't seem to find any solution for the problem that I'm having. I had taken a very similar approach as #8305 and managed to get realsense-viewer to open but it's not working properly.

I keep getting this message in terminal:
WARNING [281473097658656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

 
RealSense Viewer:
[image]

the warning message also appears in the viewer and the viewer fps is really low for 640x480
interestingly the camera works with most rs- examples like rs-align rs-enumerate-devices etc. but when I try rs-benchmark:

|            |     |
|------------|-----|
|**CPU** |unknown |
|**GPU** | V3D 4.2 |
|**Graphics Driver** |2.1 Mesa 20.2.6 |
|**Device Name** |Intel RealSense L515 |

**Stream Type**: Depth, **Resolution**: 640 x 480
Segmentation fault (core dumped)

not sure what to make of this lol
moreover, whenever I try to run python examples most of them run into a RuntimeError because of pipeline.start()

I feel as if all these issues might somehow be related so I thought it was worth mentioning them all in one place. Any help would be very appreciated!

jmmeneguel, 2021-02-25:
I know that is not your case, but I hope I can help somebody.

I installed SDK and started getting the Segmentation fault error. I reinstalled compiling from the source by the RSUSB method and the Segmentation fault error was resolved, but then the "control-transfer returned error" appeared. Later I discovered that the RSUSB method was not compatible with multicam setup.

What worked for me was installing Ubuntu 18.04.5 LTS, with kernel version 4.15.18. The only drawback I faced by now was that the confidence stream is not available in this kernel version.

RealSenseSupport, 2021-02-25:
Thanks for the information @jmmeneguel 

@Borzborz - Were you able to find a solution or configuration that helped?

Borzborz, 2021-03-05:
Thanks @jmmeneguel I will try Ubuntu 18.04.5 LTS and see if that fixes the problem

RealSenseSupport, 2021-03-23:
Hi @Borzborz 

Were you able to get past this issue?

If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-04-02 by RealSenseSupport