realsenseai/librealsense #8313: D455 drops frames in slave sync mode

sam598, 2021-02-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.12.10.0 | 
| Operating System & Version |  Raspberry Pi OS | 
| Kernel Version (Linux Only)    |  5.4.65-v7l+  | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  2.41.0  | 
| Language                            |  python  | 

I have a dozen D455 cameras, each attached to an individual Raspberry Pi 4 running a custom python capture script. Most of the cameras run fine, but one has very strange behavior when running in slave or genlock mode.

When the camera's `inter_cam_sync_mode` is set to **2** or **4** the camera will only return a couple of frames per second, at a random staccato rate. The returned frame's number will also skip, even when run using a frame queue.

None of the other cameras are experiencing this behavior. The camera sync pins are all connected with shielded twisted-pair cables and resistor/capacitor ESD protection. I have also tried the same camera on multiple Raspberry Pis, multiple USB cables, and multiple sync cables. The same behavior only happens with this D455, no matter what cable or connection is used.

The camera also runs fine with sync disabled. It even runs fine with `inter_cam_sync_mode` set to **1**, and is able to generate a signal that the other cameras can sync with. But if it receives a sync signal from another D455 camera or external source, it drops many frames.

Here is the relevant code:

```
ctx = rs.context()

devices = []
for i in range(len(ctx.devices)):
    sn = ctx.devices[i].get_info(rs.camera_info.serial_number)
    print(sn)
    devices.append(sn)
    
pipeline = rs.pipeline()

config = rs.config()

sensor = ctx.devices[0].query_sensors()[0]

sensor.set_option(rs.option.inter_cam_sync_mode, 4)

sensor.set_option(rs.option.exposure, 5000)
sensor.set_option(rs.option.depth_units, 0.001)

config.enable_device(devices[0])

sensor.set_option(rs.option.emitter_on_off, 0)
sensor.set_option(rs.option.emitter_always_on, 1)
sensor.set_option(rs.option.laser_power, 150)

config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 90)

queue = rs.frame_queue(60, keep_frames=True)

cfg = pipeline.start(config, queue)

captureRealSense = True

prevDepthFrameNum = -1

frames = None

while captureRealSense:

    try:
        frames = queue.wait_for_frame().as_frameset()
    except:
        captureRealSense = False

    if frames == None:
        continue
    
    depthFrame = frames.get_depth_frame()
    
    if depthFrame:
        depthFrameNum = depthFrame.get_frame_number()
        
        if depthFrameNum > prevDepthFrameNum:
            depthDrops = depthFrameNum - prevDepthFrameNum - 1
            if depthDrops > 0 and prevDepthFrameNum > -1:
                self.droppedDepthFrames += depthDrops
                print("DROPPED " + str(depthDrops) + " DEPTH FRAMES")

            prevDepthFrameNum = depthFrameNum

pipeline.stop()
```

My first thought is that the GPIO could be damaged. However all of the other cameras have been synced with the same cables and are running fine. Also the GPIO pins on the affected camera are still able to generate a signal, so I don't know if that makes sense.

There is also the fact that the frame queue is reporting skipped frames. That must mean that it is receiving a signal to trigger frames, but something is going wrong elsewhere.

Is it possible this is some kind of a software issue? Could a factory reset help? Or is it the hardware itself that is faulty?

sam598, 2021-02-05:
The camera has the same issue on windows with the realsense viewer and a sync signal around 30hz. It reports several of the same error:

```
Timestamp: 1612492424342.530762
Severity: Error
Description: Right MIPI error
```

[image]

Other D455 cameras run fine with this setup.

MartyG-RealSense, 2021-02-07:
Hi @sam598   Could you check in the RealSense Viewer whether first resetting the affected D455 camera in the Viewer using the **More > Hardware Reset** option in that camera's side-panel options before applying any settings to it improves its stability please?

There was also a past hardware sync case using external triggering where **Right MIPI Error** was being generated when the trigger FPS was at a slightly 'off' rate, such as 14.8 FPS instead of 15 FPS, and in that case the error did not occur when the trigger FPS was 15.  My understanding from that case is that the RealSense user was setting it to 15 but the FPS was being dragged down, causing the error.

https://community.intel.com/t5/Items-with-no-label/How-to-properly-setup-external-triggering-on-D415-and-D435/m-p/569667/highlight/true#M10018

sam598, 2021-02-08:
Thanks for the suggestions @MartyG-RealSense 

The camera has been hardware reset after each run. The issue is persistent across operating systems and sessions.

As I mentioned this happens with both the traditional intercam_sync_mode (2) as well as the genlock mode (4-255). As mentioned in the whitepaper (https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras) genlock mode supports arbitrary framerates as long as it is less than half of the target framerate, so I don't think that issue is relevant.

MartyG-RealSense, 2021-02-08:
Updated advice that I have received is that genlock mode is not currently a mature feature, and so although you are free to use it and experiment with it, the original hardware sync system (mode 1 and 2) in the **Multi-Camera configurations** white-paper document should be considered the mature 'validated' method of hardware syncing cameras.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

MartyG-RealSense, 2021-02-14:
Hi @sam598  Taking into consideration the advice in the comment above, do you require further assistance with this case please?  Thanks!

sam598, 2021-02-14:
@MartyG-RealSense thank you for getting additional advice, but as I mentioned multiple times this happens both in genlock mode as well as the "original" sync mode:

> When the camera's `inter_cam_sync_mode` is set to **2** or **4** the camera will only return a couple of frames per second, at a random staccato rate. The returned frame's number will also skip, even when run using a frame queue.

> As I mentioned this happens with both the traditional intercam_sync_mode (2) as well as the genlock mode (4-255).

> The camera also runs fine with sync disabled. It even runs fine with inter_cam_sync_mode set to 1, and is able to generate a signal that the other cameras can sync with. But if it receives a sync signal from another D455 camera or external source, it drops many frames.

The behavior also exists on Windows as well as in the RealSense viewer, so I don't think that the Python and Raspberry Pi labels are appropriate.

As a reminder of the original question this only occurs on one D455 camera, the others run fine in both modes. It seems to be a hardware issue, not a software one, and I want to see if there is some way to repair the affected device.

MartyG-RealSense, 2021-02-14:
I have not encountered another case like yours and so do not know of a solution for it.  If the only problem with the camera is that it cannot act as a Slave though and it otherwise functions normally, then using that one particular camera as the Master would avoid having to replace it.

sam598, 2021-02-14:
@MartyG-RealSense I had thought about that solution and it might be viable for some use cases, but that doesn't actually solve the issue with the camera. What happens if there are two faulty cameras in one system?

MartyG-RealSense, 2021-02-14:
Thanks very much @sam598 for your latest responses.  I have passed your feedback to Intel again for further guidance about your situation with this one particular D455.

RealSenseSupport, 2021-02-16:
Hi @sam598, are you saying you're using Raspberry Pi as trigger source? if so, it is likely the frequency just falling out of the range from the native frequency of This particular D455. This D455 might have some variation in crystal from other D455. In your case, I'd suggest to tune your Raspberry Pi's trigger frequency to match This D455 native frequency - just set This D455 sync mode=1, and tune your Raspberry Pi trigger frequency to match This D455 output frequency. Use two probes of an oscilloscope to tune the Raspberry Pi trigger. Once you figure out the matching frequency, you can use it for all other D455s.

btw, a quick way to check whether this D455 is failed or not - just use 1x good D455 as Master(sync mode=1), and use This D455 as Slave (sync mode=2), see if it is working normally.

Thanks

sam598, 2021-02-16:
> btw, a quick way to check whether this D455 is failed or not - just use 1x good D455 as Master(sync mode=1), and use This D455 as Slave (sync mode=2), see if it is working normally.

I feel like I've said this multiple times but I apologize if it hasn't been clear. This camera fails both when the camera is in sync mode 4 and receives a custom signal from a microcontroller (an Arduino as a Raspberry Pi GPIO's timing is not accurate enough) and in sync mode 2 when it receives a signal from another D455 in sync mode 1.

As I also mentioned in my original post, what is interesting is that in both setups the depth frame number is increasing with each sync signal. In an ESD event the frame number resets to 0, and when I've had a frequency mismatch I have noticed the camera just ignore the trigger and doesn't increase the frame number. This issue appears to be something different.

> This D455 might have some variation in crystal from other D455.

Does this mean that some D455 cameras will never be capable of sync? If I end up purchasing two of these faulty cameras for the same system is there a way to get a replacement?

RealSenseCustomerSupport, 2021-02-22:
----------------------------------------------

Hi sam598

"btw, a quick way to check whether this D455 is failed or not - just use 1x good D455 as Master(sync mode=1), and use This D455 as Slave (sync mode=2), see if it is working normally."
Regarding the above setup, please confirm you were using 30FPS for both Master and Slave. If you do see same issue on Windows 10 too, I guess this unit might have something physically wrong. Btw, so far I haven't seen any D435 or D455 have sync issue because of the frequency variation of the crystal. The spec says its rating is +/- 20ppm which is good enough to the camera.
Anyway, you definitely can get replacement if it is a faulty one. But I just wanted to make sure it is. Please just check it on Windows 10, and let me know the result.

Thanks

sam598, 2021-02-23:
Yes, 30FPS on Windows 10 for both Master and Slave cameras.

> Btw, so far I haven't seen any D435 or D455 have sync issue because of the frequency variation of the crystal. The spec says its rating is +/- 20ppm which is good enough to the camera.

I am curious about this. Does it mean that if a camera set to sync mode 2 has a crystal frequency of exactly 30.00hz, that the incoming sync signal from a camera in sync mode 1 must be between 29.66hz and 30.33hz?

For example:
`( ( 30fps * 60sec ) + 20ppm ) / 60sec = 30.33hz `
and
` ( ( 30fps * 60sec ) - 20ppm ) / 60sec = 29.66hz`

RealSenseCustomerSupport, 2021-02-24:
----------------------------------------------

Great, thanks. So when you test Sync Mode 1 and 2 between two D455, do you see the "Time of Arrival" is same between the two cameras? ... or after swapping the Master and Slave cam, the problem is gone?

There is a very small window for the sync trigger frequency, i.e. something like +/-40us at 30FPS. For example, if the Slave camera native frame time is 33.3333ms (30.0000Hz), the trigger range must be within 33.3733ms(29.9641Hz) ~ 33.2933ms(30.0361Hz). The window is much smaller than what you estimated.

closed 2021-03-09 by RealSenseCustomerSupport