realsenseai/librealsense #8316: Using getData() results in pixelated color frames

idenc, 2021-02-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  NA                                 | 
| Platform                                 | Android  |
| SDK Version                            |  {2.41 }                          | 
| Language                            |  {Java }                          | 
| Segment			| Smartphone                          | 

### Issue Description
Hello,

I am trying to create an Android app that performs object detection and detects the distance of each detected object. I started by using the examples [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples) and the colour and depth frames are displaying correctly. However, when I use the getData() function, convert it to a Bitmap and save it, the image is coloured poorly and is pixelated. Here is my code for getting the data:

```
private void getFrameData(FrameSet processed) {
    Frame frame = processed.first(StreamType.COLOR, StreamFormat.RGB8);
    if (frame == null) {
        LOGGER.i("Frame is null");
        return;
    }
    StreamProfile profile = frame.getProfile();
    if (profile.getType() != StreamType.COLOR || profile.getFormat() != StreamFormat.RGB8) {
        LOGGER.i("Frame has type " + profile.getType());
        LOGGER.i("Frame has format " + profile.getFormat());
        return;
    }
    LOGGER.i("frame format " + frame.getProfile().getFormat() + "\n");
    if (imageBytes == null) {
        imageBytes = new byte[frame.getDataSize()];
        LOGGER.i("Frame size " + frame.getDataSize());
        LOGGER.i("Frame format " + frame.getProfile().getFormat());
    }
    frame.getData(imageBytes);
}
```
And converting it to a Bitmap:
```
public Bitmap getIntelBitmap() {
    int size = intelSize.getHeight() * intelSize.getWidth();
    int[] pixelData = new int[size];
    for (int i = 0; i < size; i++) {
        // pack 3 bytes into int for ARGB_8888
        int r = imageBytes[3 * i];
        int g = imageBytes[3 * i + 1];
        int b = imageBytes[3 * i + 2];
        pixelData[i] = Color.rgb(r, g, b);
    }
    Bitmap map = Bitmap.createBitmap(pixelData,
            intelSize.getWidth(),
            intelSize.getHeight(),
            Bitmap.Config.ARGB_8888);
    return Bitmap.createBitmap(map, 0, 0, intelSize.getWidth(), intelSize.getHeight());
}
```
saved bitmap
[image]
app display
[image]

MartyG-RealSense, 2021-02-05:
Hi @idenc   A good starting point in this case may be to look at the Android Java scripting in the link below that creates a bitmap, also using **Bitmap.Config.ARGB_8888** like your own Java application does.

https://github.com/IntelRealSense/librealsense/issues/6089#issuecomment-602651364

idenc, 2021-02-06:
I didn't want to use OpenCV as the rest of my app does not use it and I didn't want to include it just for this one thing. The issue ended up being that Java bytes are signed by default, while createBitmap() expects unsigned bytes. Simply converting the signed bytes to unsigned bytes using the code below fixed the image.

```
for (int i = 0; i < size; i++) {
    // pack 3 bytes into int for ARGB_8888
    int r = imageBytes[3 * i] & 0xFF;
    int g = imageBytes[3 * i + 1] & 0xFF;
    int b = imageBytes[3 * i + 2] & 0xFF;
    pixelData[i] = Color.rgb(r, g, b);
}

Bitmap map = Bitmap.createBitmap(pixelData,
        intelSize.getWidth(),
        intelSize.getHeight(),
        Bitmap.Config.ARGB_8888);
```

closed 2021-02-06 by idenc

MartyG-RealSense, 2021-02-06:
Thanks very much @idenc for sharing your solution!

bjain409, 2021-03-19:
> I didn't want to use OpenCV as the rest of my app does not use it and I didn't want to include it just for this one thing. The issue ended up being that Java bytes are signed by default, while [...]

Hello @idenc 
Can you please help me with
#8551

idenc, 2021-03-19:
I'm not exactly sure what you need, but I assume you want the frame data? If so then once you have a frame just use 
```
byte[] imageBytes = new byte[frame.getDataSize()];
frame.getData(imageBytes);
```
and the frame bytes will be stored in the byte array.
Then you can use the code I posted above to convert it into a bitmap

bjain409, 2021-03-19:
> I'm not exactly sure what you need, but I assume you want the frame data? If so then once you have a frame just use
>
> ```
> byte[] imageBytes = new byte[frame.getDataSize()];
> frame.getData(imageBytes); [...]

@idenc 
Thanks for responding.
I will explain what I need.
I need functionality by which I get RGB camera data whenever I call the method.
I am creating an application that runs in background 24*7. Whenever I want to capture something,
I should get the frames of that time. The application will strictly have no UI, No UI rendering of images on display.
For instance, take an example  I have created a job in android which triggers the camera every 30 mins. So in these, I should get the data from the camera, Maybe Byte64 encoded or anything. I want that image in String or byte whichever available.
Do you have something a kind of method by which if I call that method which will return me data in form of byte or string?
I am new to android so I am not sure for Handler and message queue whether required or not as we are not needing any UI operation.
I even need a depth sensor matrix simultaneous to this. I don't require the actual image of the depth sensor.
I need metadata of individual pixel which has the depth of that pixel from a real sense device.
Whenever we open real sense application on pc after selecting depth camera, while in RGB format whenever we select a point on the image we get the distance of that point(pixel) from the depth sensor.
I am assuming the depth sensors returns just the matrix as metadata for the all pixel in the frame and then system converts that into RGB format.
Is that possible to get both camera image and depth matrix at the same time? Both thread simultaneous
Consider that I will be using a very high-end android device.

Thanks.

bjain409, 2021-04-20:
> I'm not exactly sure what you need, but I assume you want the frame data? If so then once you have a frame just use
>
> ```
> byte[] imageBytes = new byte[frame.getDataSize()];
> frame.getData(imageBytes); [...]

@idenc 
Can you please share the whole code?
I am finding it a bit difficult for getting the Frameset object.
Can you please share the whole class?
Thanks