realsenseai/librealsense #8317: Appling filters messes the color mapping when using two bag files.

AJMena, 2021-02-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.41.0.2657 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
We are doing some processes with two bag files of two D455 cameras. We load both files a synchronize the frames.
If we apply post processing filters to the frames when we do the mapping and extract the points with the color RGB, all the points of first frame have the same color, specifically the RGB value (144,130,122). 
We apply the filters like this:

			rs2::frame depth1, color1, depth2, color2;
		        rs2::disparity_transform depth_disparity(true);
		        rs2::spatial_filter spatial_filter;
		        rs2::temporal_filter temporal_filter;
		        rs2::disparity_transform disparity_depth(false);

			if (postprocessing == true && depth1 && depth2)
			{
				depth1 = depth_disparity.process(depth1);
				depth1 = spatial_filter.process(depth1);
				depth1 = temporal_filter.process(depth1);
				depth1 = disparity_depth.process(depth1);

				depth2 = depth_disparity.process(depth2);
				depth2 = spatial_filter.process(depth2);
				depth2 = temporal_filter.process(depth2);
				depth2 = disparity_depth.process(depth2);
			}

This later messes the depth color mapping result. 
If we don't apply the filters works fine. If we only use one bag file works fine.

How can we solve this?

ev-mp, 2021-02-05:
@AJMena, an immediate observation is the reuse of all the processing blocks to handle data from different sources.
Quote from [post-processing blocks documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#design-and-implementation)
>The Temporal filter effectiveness relies on the frame's history being preserved. Switching the frame sources invalidates the preserved history and incapacitates the filter. Therefore establishing and maintaining a filter pipe per camera source is strongly recommended.

Temporal filter **relies on previous samples** to denoise the image in a way that resembles running median. Feeding it with frames from two separate and distinct sources will produce incorrect results.
You need to establish separate temporal filter instances, per depth source.

AJMena, 2021-02-05:
You are absolutely correct. Already solved by using multiple sources for filters.

closed 2021-02-05 by AJMena