realsenseai/librealsense #8318: Warnings while starting realsense-viewer (patch for fourcc code RW16, relevant metadata params are missing, Invalid md size)

HartmannSa, 2021-02-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Ubuntu 18.04.5 LTS| 
| Kernel Version (Linux Only)    |  5.4.0-65-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            |  librealsense2.41.0-0                        | 
 

### Issue Description
 I get the following warnings when starting the realsense-viewer: 
```
 05/02 12:03:43,797 WARNING [140316942813056] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 05/02 12:03:54,454 WARNING [140316598310656] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing
 05/02 12:03:58,119 WARNING [140316598310656] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 05/02 12:03:58,119 WARNING [140316598310656] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 05/02 12:03:58,119 WARNING [140316598310656] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 05/02 12:03:58,119 WARNING [140316598310656] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
```
I found a similar issue #4971, however I do not use Kernel 4.16.
I have built the package from source.

MartyG-RealSense, 2021-02-05:
Hi @HartmannSa  I have seen a small cluster of cases involving an **Invalid md size: bytes used =  0 ,start offset=10** with the RealSense Viewer program during the past month and they have only occured on SDK **2.41.0**.  An example of such a case that matches closely to your particular set of errors is in the link below.

https://github.com/IntelRealSense/librealsense/issues/8154#issuecomment-761877714

So the options that I would suggest are:

1.  Try building librealsense **2.40.0** from source code; or
2. Wait for the next SDK version.  
3.  Try an **RSUSB** build of the source code and see whether using this installation method (which is not dependent on kernel versions and does not require patching) makes a positive difference to your installation.

HartmannSa, 2021-02-05:
Okay thanks.
So, I would just do `sudo make uninstall` and then delete the folder.
And then download the correct version from here https://github.com/IntelRealSense/librealsense/releases and build it again.

Or is there an easy/ better way to downgrade?

MartyG-RealSense, 2021-02-05:
It would make logicial sense to me to download the source code folder for **2.40.0** and go through the usual process of **mkdir build && cd build** to create the Build folder.  And then run the full instrution for uninstall, clean-up and install as stated by the source-code installation documentation:

**sudo make uninstall && make clean && make && sudo make install**

HartmannSa, 2021-02-05:
okay, thanks again for your fast reply!

The warning **Invalid md size: bytes used = 0 ,start offset=10** doesnt show up with version 2.40.0.

The other warnings still occur (whereby the second warning only occurs when the depth stream is activated):
```
 05/02 20:01:39,339 WARNING [140204514723584] (backend-v4l2.cpp:1271) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 05/02 20:01:50,161 WARNING [140204354557696] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing
```

Do you know something about them?

MartyG-RealSense, 2021-02-05:
When the **fourcc code RW16!** message occurs, it would suggest a conflict with the Linux kernel.  You could test this possibility by building librealsense from source code using the **RSUSB** installation method, which is not dependent on Linux versions or kernel versions and does not require patching.

HartmannSa, 2021-02-05:
Can you explain what the implications of this warning are if I don't fix it?
I don't understand what the error message is trying to tell me.
Since everything else is currently working, I don't know if I should try a new build...

MartyG-RealSense, 2021-02-05:
RealSense team member @ev-mp can provide a better technical explanation of **fourcc code RW16!** than I can.  Below are links to a couple of previous comments by @ev-mp that make reference to the kernel patching topic though.

https://github.com/IntelRealSense/librealsense/issues/4971#issuecomment-537649983
https://github.com/IntelRealSense/librealsense/issues/3686#issuecomment-515775903

It is of course your choice whether you build again or not.  It may come down to how much the error affects your use of librealsense overall.

HartmannSa, 2021-02-06:
Thanks, I have read the comments. But they did not really help me. 
If I understand correctly, the answers refer to the "problematic" kernel 4.16 (I have 5.4) and the realsense2-dkms package (which I don't use because of my installation from source).

MartyG-RealSense, 2021-02-06:
The two links - which also feature **fourcc code** errors - were intended to be a general example of the problems that kernel conflicts can cause if they are not fixed.  This was in reply to your question above of "Can you explain what the implications of this warning are if I don't fix it?"

@ev-mp would be better able than myself to provide information that relates specifically to the **fourcc code** error in your own case.

MartyG-RealSense, 2021-02-12:
Hi @HartmannSa  Do you require further assistance with this case, please?  Thanks!

HartmannSa, 2021-02-13:
Hi, since the warnings do not currently interfere with my program, I simply ignore them. 
However, I was hoping that @ev-mp would answer my questions: 
> Can you explain what the implications of this warning are if I don't fix it?
> I don't understand what the error message is trying to tell me.
> Since everything else is currently working, I don't know if I should try a new build...

MartyG-RealSense, 2021-02-13:
@HartmannSa   Given that the RealSense user in the recent case that was the closest match to your set of errors in https://github.com/IntelRealSense/librealsense/issues/8154#issuecomment-767175826 was given the advice that the next librealsense version after **2.41.0** should fix that case's errors, I would recommend waiting for the next SDK version to see if the changes in that version resolve your problem too.

closed 2021-02-15 by HartmannSa