realsenseai/librealsense #8319: L515 Resolution limited

LarsBuck, 2021-02-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.03.00 | 
| Operating System & Version |   Ubuntu 20.04 LTS | 
| Kernel Version (Linux Only)    |  5.4.0-65                                        | 
| Platform                                 | PC|
| SDK Version                            |  2.41.0                          | 
| Language                            |  python  | 
| Segment			|  Robot                          | 

Hello,

So i know that the Realsense L515 RGB resolution is 1920 x 1080 and depth resolution is up to 1280 x 720. However im only getting 640 x 480 for RGB and 320 x 240 for Depth. Im pretty sure that the Problem comes along with the fact, that the Camera gets recognized as USB 2.1 although the camera is plugged into an usb3 port with the cable that comes with the camera. I know the USB problem is known and i already tried all the solutions you mentioned on you support side. Everything else is working fine, i allready started working with the lib and enjoy the cam but its sad that i cant get the full resolution working.
down below is the device info that confirms my problem…

thx for your time

Device info: 
    Name                          :     Intel RealSense L515
    Serial Number                 :     f0270316
    Firmware Version              :     01.05.03.00
    Recommended Firmware Version  :     01.05.03.00
    Physical Port                 :     /sys/devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/video4linux/video0
    Debug Op Code                 :     15
    Product Id                    :     0B64
    Camera Locked                 :     YES
    Usb Type Descriptor           :     2.1             <--- ???
    Product Line                  :     L500
    Asic Serial Number            :     0003ac80081c
    Firmware Update Id            :     0003ac80081c

Stream Profiles supported by L500 Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Confidence    320x240       @ 30Hz     RAW8
    Infrared      320x240       @ 30Hz     Y8
    Depth         320x240       @ 30Hz     Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color         640x480       @ 30Hz     RGB8
    Color         640x480       @ 30Hz     Y16
    Color         640x480       @ 30Hz     BGRA8
    Color         640x480       @ 30Hz     RGBA8
    Color         640x480       @ 30Hz     BGR8
    Color         640x480       @ 30Hz     YUYV
    Color         640x480       @ 15Hz     RGB8
    Color         640x480       @ 15Hz     Y16
    Color         640x480       @ 15Hz     BGRA8
    Color         640x480       @ 15Hz     RGBA8
    Color         640x480       @ 15Hz     BGR8
    Color         640x480       @ 15Hz     YUYV
    Color         640x480       @ 6Hz      RGB8
    Color         640x480       @ 6Hz      Y16
    Color         640x480       @ 6Hz      BGRA8
    Color         640x480       @ 6Hz      RGBA8
    Color         640x480       @ 6Hz      BGR8
    Color         640x480       @ 6Hz      YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel        N/A            @ 400Hz    MOTION_XYZ32F
    Accel        N/A            @ 200Hz    MOTION_XYZ32F
    Accel        N/A            @ 100Hz    MOTION_XYZ32F
    Gyro         N/A            @ 400Hz    MOTION_XYZ32F
    Gyro         N/A            @ 200Hz    MOTION_XYZ32F
    Gyro         N/A            @ 100Hz    MOTION_XYZ32F

LarsBuck, 2021-02-05:
Update: so i just received another l515 today, which works perfectly fine with both usb3.0 ports. So i assume the problem lays with the camera chip. Any other ideas whats possibly wrong ? 

Device Info of the second camera: 

Device info: 
    Name                          :     Intel RealSense L515
    Serial Number                 :     f0246105
    Firmware Version              :     01.05.03.00
    Recommended Firmware Version  :     01.05.03.00
    Physical Port                 :     /sys/devices/pci0000:00/0000:00:14.0/usb4/4-5/4-5:1.0/video4linux/video8
    Debug Op Code                 :     15
    Product Id                    :     0B64
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     L500
    Asic Serial Number            :     0003ac826d3c
    Firmware Update Id            :     0003ac826d3c

RealSenseSupport, 2021-02-10:
Hi,

If you plug both L515s into another system, do you also experience the same behavior (USB2 on one and USB3 on the other?) If so it may indeed be an issue with the camera so feel free to RMA it back..

Thanks

LarsBuck, 2021-02-11:
Hello,

thanks for your response, i will give that a try on the weekend. I will inform you back then.

Greets
Lars

LarsBuck, 2021-02-17:
Hi,

just wanted to quickly update and close.
Unfortunetly nothing worked out to fix the USB recognition.

closed 2021-02-17 by LarsBuck

wangziqi000, 2021-03-24:
Hi,

I experienced the same problem as @LarsBuck. After updating the firmware to the new version recommended in realsense-viewer, the device is then recognized as a USB2.0 device and the only available resolution is 320*240 for depth and 640*480  for rgb.  Very weird. I revert the firmware to 1.05.02.00 and it worked as before.

You might want to check if there are problems with the new firmware. @RealSenseSupport 
Thanks for any help!