realsenseai/librealsense #8320: Add option/code to output imu/motion data to rs-convert

jimmek, 2021-02-05:
| Camera    | D455           |
| Firmware | 5.12.10.00   |
| OS           | Win 10        |
| SDK         | 2.41.0.2657 |

### Issue Description
Having an option in rs-convert to output imu/motion data to an ASCII file would be a welcome addition to the program based on the many issues regarding access to imu/motion data that I have seen o GitHub/IntelRealSense/librealsense.  Given that the realsense-viewer displays the accel and gyro data on screen as it is collecting, could not the code in viewer that generates the values be modified and incorporated within rs-convert as there is imu/motion data in the bag file.?

MartyG-RealSense, 2021-02-05:
Hi @jimmek   The SDK tool **rs-data-collect** can store and serialize D435i IMU data to a text-based **.csv** file that is compatible with spreadsheet software such as Excel.  

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect

The documentation does not refer to D455 but as the program seems to auto-configure the data collected based on the hardware features of the detected camera model, it may work similarly for the IMU-equipped D455 as it does on D435i.

In the Windows version of the SDK that is installed by the installer file from the **Assets** file list on the SDK **Releases** page, a pre-built executable of **rs-data-collect** can be found in the SDK's **Tools** folder (the same one where **rs-convert** can be found).  It can be launched in the Windows Command Prompt using a textual command input (described in the link above).

jimmek, 2021-02-08:
Thank you for the reference to rs-data-collect. I did run it to see what the output looked like. I was able to see gyro data output but got an error messages regarding the accelerometer. Here is the command line,  configuration file content, and a screenshot of the Accel error message for the D455. 

rs-data-collect.exe -c data-collect.cfg -f rs-data-collect.out -t 10 -m 1000

ACCEL,0,0,62,MOTION_XYZ32F
GYRO,0,0,200,MOTION_XYZ32F

[image]

Where I had previously suggested that portions of the viewer code be incorporated into an enhanced rs-convert, it would seem more appropriate to use a portion of the rs-data-collect code within rs-convert. Program rs-data-collect is more of demonstration program where data collection is limited to a time or frame limit that assumes this is know for the data collection situation. An enhanced rs-convert that reads bag files would not have this limit and thus be more flexible in the application of the IMU data.

MartyG-RealSense, 2021-02-08:
Hi @jimmek   I note that RealSense team member @ev-mp has added an **Enhancement** tag (i.e a feature request) to this case.  So @ev-mp would be best able to comment on the possibility of IMU data logging in a csv file being added to **rs-convert** in the future.

jimmek, 2021-02-08:
In looking at the bag files from rs-data-collect, I can see the references to the Accelerometer data but not the Gyro data. Is it written to the bag file?   In also looking at viewer generator bag files, I also do not see a reference to the gyro. Can you check into this?

jimmek, 2021-02-08:
My above message is confusing. Sorry. Let me restate in an easy to reply manner.  Can you confirm that both Accel and Gyro files are written to the bag files from rs-data-collect and viewer program? As you can see from above, I got gyro data from rs-data-collect but not accel data due to possible cfg file error. In another bag file output by viewer, I do see Geometry_msgs/ Vector3 and Quaternion.

MartyG-RealSense, 2021-02-08:
No problem  :)   **rs-data-collect** writes IMU data to text-based **csv** file.  The **RealSense Viewer** can write IMU data (both gyro and accel) to a **bag** file.

The SDK **Sample Data** page has a selection of sample bag files as downloadable zips.  It has an **outdoor scene** zip that contains two bags that contain gyro and accel data.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

[image]

jimmek, 2021-02-08:
MartyG
In regard to the above comment on the error in running rs-data-collect as shown in the **_console_** screen capture, the two line configuration file content for the ACCEL and GYRO streams, and the command line do you see anything wrong that would have generate the error message? I used the samples in the rs-data-collect document as the guide for the configuration file content. Output file did have gyro data and no ACCEL data.

rs-data-collect.exe -c data-collect.cfg -f rs-data-collect.out -t 10 -m 1000

ACCEL,0,0,62,MOTION_XYZ32F
GYRO,0,0,200,MOTION_XYZ32F

jimmek, 2021-02-08:
Hello Marty,

Resolved problem getting ACCEL data. Sample rate should be 63 not 62.

MartyG-RealSense, 2021-02-08:
Thanks very much for the update of your success @jimmek !

jimmek, 2021-02-12:
Still looking for responses regarding enhancement suggestion for rs-convert to also output IMU data.  Are there any other users of rs-convert that would want this enhancement added to rs-convert?

MartyG-RealSense, 2021-03-03:
Hi @jimmek   An official feature request has been filed with Intel for the addition to **rs-convert** of support for IMU information extraction.  There is not a time estimate that can be provided regarding implementation of this request though.  This case will be kept open whilst the feature request is active.

jimmek, 2021-03-04:
Thank you MartyG.  I think the users of the RealSense cameras with IMU will appreciate this feature as it provides a complete picture of what is contained in the BAG file.

MartyG-RealSense, 2021-06-20:
Hi @jimmek   Support for serializing IMU and Pose data into rs-convert was added in SDK version **2.47.0**.

[image]

Though the feature has been released, please keep this case open for a further time period.

MartyG-RealSense, 2021-06-26:
This case is now able to be closed.  Thanks again @jimmek   :)

closed 2021-06-26 by MartyG-RealSense