realsenseai/librealsense #8321: Can

brizio73, 2021-02-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 and D435 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version | Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  5.4.0-65-generic                                        | 
| Platform                                 | PC and Raspberry Pi  |
| SDK Version                            | 2.40 and 2.41                         | 
| Language                            |  C                         | 
| Segment			|  application                         | 

### Issue Description

The short story: it seems that the 'owner' pointer inside the frame can be NULL. When this happens, there are cases in which the pointer needs to be referenced, but being the pointer NULL, the application crashes.

The complete story: I discussed it for long time on the community forum, here: [https://support.intelrealsense.com/hc/en-us/community/posts/360052493073-Post-processing-filters-in-C-leak-and-segmentation-fault?page=2#community_comment_1500000287222](url)

Best regards!

ev-mp, 2021-02-09:
@brizio73 hello,
One common pitfall is that rs2::frame objects are ref-counted, thus trying to obtain and handle to the underlying memory object directly will create inconsistency between the object's state and the allocated resources.
It would be useful to share some code snippet of the frames processing flow

brizio73, 2021-02-09:
Hi @ev-mp, thanks for your response. I was expecting that rs2::frame was ref-counted, but the problem arise in a fairly simple piece of code. This is an extract of the code:
```
        // read RealSense frames
        rs2_error *err = NULL;
        // gets composite frames (may contain more than one frame, for different sensors also)
        rs2_frame *frames = rs2_pipeline_wait_for_frames(input_streams->rs_pipeline, RS2_DEFAULT_TIMEOUT, &err);
        int num_of_frames = rs2_embedded_frames_count(frames, &err);
        int i = 0;
        for (i = 0; i < num_of_frames; ++i)
        {
            // for each frame in the composite, check if it is a depth frame
            rs2_frame* depth_frame = rs2_extract_frame(frames, i, &err);
            if (err != NULL || depth_frame == NULL)
            {
                if (depth_frame != NULL) rs2_release_frame(depth_frame);
                if (err != NULL) rs2_free_error(err);
                continue;
            }
            if (0 == rs2_is_frame_extendable_to(depth_frame, RS2_EXTENSION_DEPTH_FRAME, &err))
            {
                // if not, keep going
                if (depth_frame != NULL) rs2_release_frame(depth_frame);
                continue;
            }

            if (input_streams->rs_colorizer_filter == NULL)
            {
                // log: "[!!!] This should never happen! where's my colorizer filter???"
                continue;
            }
            rs2_process_frame(input_streams->rs_colorizer_filter, depth_frame, &err);
            if (err) { macap_logger_debug("ERROR: %s\n", rs2_get_error_message(err)); break; }
            rs2_frame *colorized_frame = rs2_wait_for_frame(input_streams->rs_colorizer_queue, RS2_DEFAULT_TIMEOUT, &err);
            if (err || colorized_frame == NULL) { /* log: "ERROR" */ break; }

            // depth frame, at last! get the data
            const uint8_t *depth_frame_data = (const uint8_t*)(rs2_get_frame_data(colorized_frame, &err));
            if (err == NULL)
            {
                int depth_stride = rs2_get_frame_stride_in_bytes(colorized_frame, &err);
                int depth_size = rs2_get_frame_data_size(colorized_frame, &err);
                int depth_w = rs2_get_frame_width(colorized_frame, NULL);
                int depth_h = rs2_get_frame_height(colorized_frame, NULL);

                fullsize_img.stride = _videosource->stride;
                fullsize_img.width = _videosource->width;
                fullsize_img.height = _videosource->height;
                fullsize_img.size = _videosource->size;
                fullsize_img.buffer = (unsigned char*) rapp_malloc(fullsize_img.size, 0);
                int row = 0, col = 0;
                for (row = 0; row < depth_h; ++row)
                {
                    for (col = 0; col < depth_w; ++col)
                    {
                        // when color is provided by colorizer block:
                        int src_offset = row * depth_stride + (col*3);
                        int dst_offset = row * _videosource->stride + col;
                        uint8_t pixel = depth_frame_data[src_offset];
                        fullsize_img.buffer[dst_offset] = pixel;
                    }
                }
                fullsize_img.data = fullsize_img.buffer;
                fullsize_img.padding = 0;
                fullsize_img.binoffset = 0;
                fullsize_img.disposable = true;

```
The crash happens (sometimes! not a deterministic behavior!) on the call to rs2_process_frame(). I explored it with valgrind and this is the last call stack I get:

[image]

(sorry for posting an image, I had the valgrind report in xml format which is a little verbose to read in plain text). Please note the innermost call to librealsense::frame::release() and the line marked with the icon (which is the instruction that caused the segfault).

I have been struggling on this issue for months now. I always supposed it was my fault, maybe I am misusing the API in some way (and I still hope so!) but I really can't find any evident error on my side, even comparing the code to the examples.
Any help will be highly appreciated!

best
Fabrizio

brizio73, 2021-02-09:
Ah let me add some comments.
The code is supposed to grab a depth frame, turn it into a standard RGB image and copy it in a custom structure, here called 'fullsize_img'. The copy is really naive and probably not very efficient, but it's the first attempt. It should work in any case, and ideed it does, until the segfault arise.

The code is not complete, but I cut it for briefness. However you can consider the last 'if' statement and the 'for' loop to end right after my snippet. I'll see if I can edit the previous post...

ev-mp, 2021-02-09:
@brizio73 , thanks for the example code.
Actually, you're using C-API where the issue can be the opposite as there is no native RAII in C. (no dtors/automatic deleters)
So each resource allocated using the C-API  (frame, pipe, post-processing filter) must be tracked and released manually in code at the end of usage/scope immediately.
```cpp
          rs2_frame* depth_frame = rs2_extract_frame(frames, i, &err);
... 
          rs2_release_frame(.. call is envisaged.
           
```
Please take a look at https://github.com/IntelRealSense/librealsense/issues/4006#issuecomment-494033057 where a similar discussion took place

brizio73, 2021-02-09:
Sorry, after all I cut off a relevant piece of code from my snippet above. At the end I did the following:
```
            // other irrelvant code (not rs2 related)

            // cleanup (this in case you haven't broke the previous for loop)
            rs2_release_frame(depth_frame);
            rs2_release_frame(colorized_frame);

            // in any case, break after the first depth frame
            break;
        }   // this closes the if statement

        rs2_release_frame(frames);
        if (err)
        {
            rs2_free_error(err);
        }
    }   // this closes the for loop

```

RealSenseSupport, 2021-02-18:
@brizio73, please let us know if any further help needed for this case. Thanks.

brizio73, 2021-02-19:
Hi @RealSenseSupport , thanks for following up. So far there isn't any news, the code still behaves the same, crashing after a while. I saw the discussion linked by @ev-mp but that was a different case, I free all the allocated memory in my main loop (originally I forgot to copy the last part of the loop, see my previous comment). Unfortunately I cannot work 100% of my time on that topic, so I could not dig deeper in the issue in the last few weeks. Reviewing the conversation now, I wonder if it could be possible that I am freeing "too much" memory... Maybe I am freeing a pointer when I should not?

Another thing I'd like to point out is that before posting here, I made quite an extensive use of Valdgrind trying to figure out what the problem is, and the Valgrind report is full of potential errors. Is that known to you or even desired? This happens with the sample code also. It would be great if all this warnings could be removed. Most of them are "Conditional jump or move depends on uninitialised value(s)". If these warnings cannot be removed, a "valgrind ignore list" would be really helpful...

Thanks a lot for any further help you will be able to provide!

workingenius, 2021-02-26:
I guess we are facing the same error.  @brizio73 Did you find a solution to it? thanks.

Required Info |  
-- | --
Camera Model | D410
Firmware Version | Unknown
Operating System & Version | Android 8.1
Kernel Version (Linux Only) | Linux version 4.4.167 (mengpengfei@apr-11)
SDK Version | 2.41
Language | C++/Java
Segment | application

[image]

RealSenseSupport, 2021-05-06:
The crash dump seems to be similar to #5424. Please check and see if the issue can be resolved. 
Thank you.

workingenius, 2021-05-07:
@RealSenseSupport yeah you are right, I guess they are the same problem, according to the call stack.

But #5424 didn't give a solution, I don't know what to do then.

Someone else is asking the same question. I read [this](https://support.intelrealsense.com/hc/en-us/community/posts/360052493073-Post-processing-filters-in-C-leak-and-segmentation-fault?page=2#comments). 

Why `owner` is null ? It is constructed as null, or someone else has already released it ?

I'm trying to go deeper for the problem. But now I have difficulties reading the source codes. I can't tell the difference between `librealsense` and `librs`, and there seems to be more than one class named `frame` in the whole repository. So it's harder for me to find out where the frames are possibly constructed. It's helpful if you have some guide or documents on navigating the source code, maybe the purpose of each part, each class etc. I'm not an expert though, but I'm grateful for just showing me where to start.

workingenius, 2021-05-07:
Oh, I realize maybe it's @brizio73 who did the research in [the article](https://support.intelrealsense.com/hc/en-us/community/posts/360052493073-Post-processing-filters-in-C-leak-and-segmentation-fault?page=2#comments) ?

brizio73, 2021-05-07:
Hi everybody!

@shangruqi, yes, it's me who wrote "the article". I found the problem a few
months ago while trying to integrate the RealSense depth camera in a
product from my company but that error forced us to stop. I was then forced
to move to another project and could not investigate deeper, but I am
really glad some else is investigating it. Here's a few considerations I
made lately...

   1. although my analysis clearly pointed out that the problem is the
   dereferenciation of a null pointer, in many cases this does not happen. I
   tried to reproduce the bug with a different (simpler) code, and never
   succeeded. So why is this happening? I still suspect that we (I) may have a
   responsibility in this issue and that maybe we could just solve it by using
   the API in a different way (not easy though: my existing code is a quite
   complex multithreading code and cannot be simplified...)
   2. a quick workaround we could attempt is to check for the owner pointer
   before doing anything with the frame, and drop it if it is found null.
   However, I haven't had the chance to try that and I don't know if we have
   visibility of that pointer (or of the function get_owner()). Still, I'd
   like to try.

Come on, let's solve this. I am eager to read your findings!

Best
Fabrizio

workingenius, 2021-05-17:
Hi everyone.

I read some more code from master branch (commit 8f355d5e8), and I have some questions.

Firstly it's about `frame` class from `src/archive.h`, each of which has a `ref_count` and an `owner` as member. They are initialized to zero and `nullptr` respectively. `frame` can only be moved but not copied. On moving, both `ref_count` and `owner` is passed to the other object and the original values are reset and not used any longer. When `ref_count` count down to zero, some cleaning work (also involving `owner`) is triggered.

But `release` method didn't consider the situation when `owner` is `nullptr`.  That is to say, a simple snippet would crash:

```cpp
auto f = new frame();
f.acquire();
f.release();
```

**I think this is a "flaw" and should be fixed, although the class is working fine for most of times. But it crashes easily on "careless" use of it.**

More on this topic, if you check `src/source.cpp` line 117, it also implies that `owner` is not null, which cause another kind of crash sometimes. While in the move constructor of `frame` we can see that `owner` to be `nullptr` is checked. So I concluded that it is allowed to be `nullptr`, but usually it is not. So I turned to look for where the `owner` is set.

The only place where `owner` is given a value in `frame::publish` method. It is called in class `frame_archive`, a manager class that in charge of the lifetime of a `frame`. It creates `frame`s, manages memory, and maybe recycle of them. The `owner` of `frame` just points to an archive, indicating where it was created and by whom it is managed. From `frame_archive::alloc_and_track` we see that a newly allocated `frame` is bound to an archive shortly after its creation, and the `frame` is not published yet then. So I think it's not probable that the `nullptr` crash happens because of the short gap between `frame` creation and given an `owner`.

Then another idea came to me when I got aware of recycling of frames. There might be some mistakes if a `frame` has already been recycled but someone is still holding a pointer to the address. In my opinion, `frame_holder` is essentially a pointer to a `frame` with something facilitates its usage.

**Do you think the crash is caused by recycling and an incorrect pointer? Is that possible?**

As for `frame_holder`, I have some other questions. **In this constructor `frame_holder(frame_interface* f)`, the frame is not `acquire`d, but in the destructor, the frame is `release`d ? Won't there be an imbalance ?** But when a `frame_holder` is cloned, the `acquire` is called. I'm confused by the code and like to figure out how it works correctly. 

And in the end, **why bother designing a `ref_count` and `frame_holder` (as a pointer) , why not use `shared_ptr` ?**

---

My thinkings here are certainly not 100% correct, half analyzing + half guessing maybe. 
But maybe that reflect how "people" will think during purely reading the codes.
Please tell my if there's something I missed, especially if I misunderstood some c++ syntax. 
Later I'll try to build and run it, change some code and play with it.

Thank you.

RealSenseSupport, 2021-06-03:
Hi @ev-mp, please would it be possible to provide some additional advice about in this case? Thank you.

closed 2021-08-10 by RealSenseSupport