realsenseai/librealsense #8323: L515 and D455 IMU not available for MacOS in the current SDK version

RaduCorcodel, 2021-02-06:
| Required Info                         |                                     |
|---------------------------------|---------------------- |
| Camera Model                       |L515| 
| Firmware Version                   | 01.05.03.00 | 
| Operating System & Version |  MacOS  Big Sur 11.1 | 
| Kernel Version (Linux Only)    |      | 
| Platform                                 | PC  |
| SDK Version                            |  2.41.0   | 
| Language                            |  C                          | 
| Segment			|  others                         | 

| Required Info                         |                                     |
|---------------------------------|---------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   MacOS  Big Sur 11.1  | 
| Kernel Version (Linux Only)    |                | 
| Platform                                 | PC  |
| SDK Version                            |  2.41.0 | 
| Language                            |  C   | 
| Segment			|  others    | 

### Issue Description
I updated the SDK and updated the firmware on both cameras and I see now that the IMU is disabled for MacOS on both cameras. I need to use the accelerometer data and was wondering when are there plans to support the onboard IMU on these cameras? BTW, I build with 
```cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false```

RaduCorcodel, 2021-02-08:
Even a ballpark idea is greatly appreciated. We need this functionality for our system and we need to make a decision on what to do next.

ev-mp, 2021-02-09:
@RaduCorcodel , the support for IMU for MacOS was recently implemented and has already integrated into the `development` branch in #8029,  destined for the next release.
You can try it from there - feedbacks will be appreciated.

RealSenseSupport, 2021-05-25:
@RaduCorcodel Did you get chance to try the latest SDK version? Looking forward to your feedback. Thanks!

RealSenseSupport, 2021-06-02:
@RaduCorcodel Could you please update? Please note that the issue will be closed if we still don't get your update in another 7 days.

RealSenseSupport, 2021-06-09:
@RaduCorcodel The feature has been implemented. As we don't hear from you for weeks, will close this ticket at this point. If you have any other questions or issues, please feel free to create another new ticket. Thanks!

closed 2021-06-09 by RealSenseSupport