realsenseai/librealsense #8324: @MartyG-RealSense That did the trick.

josh-github1, 2021-02-07:
@MartyG-RealSense That did the trick. 

For future reference, here is a complete list of steps I took to get the IntelRealsense Camara D455 to work with python3 on the Jetson Nano. 

1. Update CMake using [this method](https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977)
(You need to have curl [installed](https://stackoverflow.com/questions/56941778/cmake-use-system-curl-is-on-but-a-curl-is-not-found) for this to work. )

2. Download the zip file from https://github.com/IntelRealSense/librealsense/releases/. I am using version 2.38.1

3. Extract the file, cd into the extracted file .

4. Create a dir called build and cd into it.  

5. Run the CMake command to test to see if the build will work.
 `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true`

If you have issues with the above step, check out [this](https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13), [this](https://github.com/IntelRealSense/librealsense/issues/6449#issuecomment-650793097) and [this](https://askubuntu.com/questions/620601/cant-install-libsdl2-dev-or-cmake-glfw-libxinerama-dev-problem)

6.  Still in the build dir. Run` make -j4` and then `sudo make install`.

7. Add these to the end of your .bashrc file
```
export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2
```

8. source your .bashrc file

9. Deal with package restructuring between python2 and python3 using [this method](https://github.com/IntelRealSense/meta-intel-realsense/issues/20#issuecomment-687180484)

You should now be able to import` pyrealsense2` using python3 

@MartyG-RealSense thank you once again for your help 💯

Update: I can confirm that this works for the Jetson Xavier NX as well.

_Originally posted by @Drkstr in https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049_

josh-github1, 2021-02-07:
I've followed all these steps and it still isn't recognizing pyrealsense2 in the python import. I got up to number 8. 

In /usr/lib/python3.6/site-packages there is no pyrealsense2 folder. 

I'll try doing the cmake and make commands and seeing how it goes but wanted to post this here in case someone is running into the same issue. This experiment is being done on the original jetson nano running on the latest jetpack and ubuntu 20. I have python3.8 also installed and sure that might be an issue but idk its late.

MartyG-RealSense, 2021-02-07:
Hi @josh-github1  May I begin by asking whether librealsense and the **RealSense Viewer** program are working correctly after installation has completed?

Also, is the information in the link below by another Nano user of this installation guide helpful to your problem?

https://github.com/IntelRealSense/librealsense/issues/7775#issuecomment-730685749

josh-github1, 2021-02-07:
> There was a Jetson case involving **AttributeError: module 'pyrealsense2' has no attribute 'pipeline'** where the solution in that particular case was found to be to copy **.so** files to the same directory that the user's project script was in.
> 
> [#7078 (comment)](https://github.com/IntelRealSense/librealsense/issues/7078#issuecomment-672949007)
> 
> This same approach of copying the .so files into the same directory as the user's project script also worked in another case:
> 
> https://support.intelrealsense.com/hc/en-us/community/posts/360048396854/comments/360012486393

> There was a Jetson case involving **AttributeError: module 'pyrealsense2' has no attribute 'pipeline'** where the solution in that particular case was found to be to copy **.so** files to the same directory that the user's project script was in.
> 
> [#7078 (comment)](https://github.com/IntelRealSense/librealsense/issues/7078#issuecomment-672949007)
> 
> This same approach of copying the .so files into the same directory as the user's project script also worked in another case:
> 
> https://support.intelrealsense.com/hc/en-us/community/posts/360048396854/comments/360012486393

MartyG-RealSense, 2021-02-07:
Excellent news that you found a method that worked for you.  Thanks for the update!

closed 2021-02-20 by MartyG-RealSense