realsenseai/librealsense #8325: Distortion parameters are all 0 for the RAW data of the RGB camera in D415

JianWang-CMU, 2021-02-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | C                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I get the intrinsic parameters of the RAW data of the RGB camera. The distortion coefficients are all 0, which means the lens doesn't have any distortion at all.
How come is that possible?
Well, I do observe that the straight line in the edges are still straight. But can someone explain that how a practical lens doesn't have any distortion. Thanks!
The only explanation that I have is, this RAW data is not the real RAW data; it is the re-sampled mosaiced data from the rectified image. Is this guess correct?

MartyG-RealSense, 2021-02-08:
Hi @JianWang-CMU   This is not an error.  The 400 Series deliberately sets all coefficients to zero.  More information about this can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/1430

The RealSense D455 model has an exception to this rule where the **Color** coefficients may be reported as being non-zero, but this does not affect your D415.

https://github.com/IntelRealSense/realsense-ros/issues/1580

JianWang-CMU, 2021-02-08:
Thanks! Those two links solved my question.
Also, in my case, I am using RAW. The RAW also has no distortion.

I am surprised that the lens has very little distortion (I don't have optics background).

MartyG-RealSense, 2021-02-15:
Hi @JianWang-CMU  Do you require further assistance with this case, please?   Thanks!

JianWang-CMU, 2021-02-15:
I don't need further assistance. Thanks!

closed 2021-02-15 by JianWang-CMU