realsenseai/librealsense #8326: "Spotty/Patchy" noise effect in L515

Faaux, 2021-02-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.02.00 | 
| Operating System & Version |  Win10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.40.0 | 
| Language                            |  {C/C++ }                          | 
| Segment			|  AR   | 

### Issue Description
Hi everyone,
I am currently trying to use the L515 to detect near surface interactions of a user's hand. For this I build a basic background model to compare against consisting of an average+std-dev of 5s of data at 1024x768@30 from the LiDAR. I tried a working distance between 1.1m and 2.5m.

With the background in place I capture another depth frame of the same scene without changing anything and compare the newly acquired depth values to the previously (meaning a couple seconds beforehand) acquired background.

The issue (which is shown in the attached image) is that a major portion of the depth values is very noisy compared to the acquire background. Furthermore, the noise can not be described as white noise since it is happening in various spots rather than uniformly across the image. 

The scene is from a white wall in my office with no lights turned on. There are windows where sunlight can enter.

Image legend: 
- Black:  Within 1 StdDev in front of BG
- Dark Grey: Within 1-3 StdDev from BG
- Light Grey: More than 3 StdDev from BG
[image]
Some pixels are also further away then the background by more than 2 StdDev but are not shown in this visualization.

Interpretation: Noise Level could be fine: 1 StdDev is ~68% of all pixels. But why is it patchy/spotty

Questions:
- Where doe the patches/spots come from?
- Is this a known physical effect? 
- Are there other background models better suited for LiDAR data?

Thanks for all the work you guys do!

Best,
Faaux

Faaux, 2021-02-08:
Looking at the noise distribution it works out, meaning it fits the expected percentage. The weird issue for me is the "spottiness" of the whole noise.

Faaux, 2021-02-09:
Another day and more information.

I was able to reproduce the test in a completely dark room with lights on and off. All scenarios show the same behavior.

RealSenseSupport, 2021-02-24:
Hi @Faaux 

What specific settings, presets, etc for this setup/configuration?

You ask about background model, I'm trying to better understand what you are trying to accomplish.

You taking depth frames over a 5s interval and averaging the depth data, why add std-dev?

Faaux, 2021-03-02:
Hi @RealSenseSupport,
I am using the "Short Range" preset with no additional modifications. Resolution can be 640x480 or 1024x768, it does not change the observed effect.

The goal for me is to observe near surface interactions of object and hands. For this I build a background model, modelled as mean and std-dev for each depth pixel over period of time. I do this to estimate the noise level, which later on is used to determine if something is close to the surface and give an accuracy estimation. If I see that I have 9mm std-dev I can for example invalidate this pixel or react appropriately.

During this testing I found that the noise pattern seen above is observed. I was assuming to see something like white-noise, maybe even speckle or other noise patterns. When I saw these "Spots" I was wondering where they come from because I cant explain them and not model them either.

Spots are especially dangerous for me in the following scenario:

Estimating how far a finger is held above the surface. If a random "spot" lies on the finger the estimated depth will definitely be wrong by >1 std-dev. I also cant average the finger depth because it is a whole "spot" across the finger and not individual fingers pixels which are noisy. Naively I would have assumed white noise which can be averaged out when taking enough pixels into account. With regions of noise this is not as simple.

Since I cannot explain the "spot" phenomenon its hard for me to offer an explanation of why this is happening or how I could model it. Are there internal averaging mechanism at different resolutions or other assumptions that could lead to averaged regions? Maybe detector changes depending on the amount of reflected light that change the depth measurement in a specific region? I am assuming that it is internal L515 behavior because the same effect is seen in a (nearly) completely dark room (using short range preset)

RealSenseSupport, 2021-03-23:
Thanks for the explanation of your setup.  

I think may be environmental related or reflectivity related.  The spots look like our the device isn't getting IR reflection from those spots.  If there is ambient light in the room or region or if the reflectivity of the hand, object or table/wall is not high. 
We do have in the Depth Quality Tool the ability to check the reflectivity of an object or scene.  There is also the option within the RealSense Viewer to check the max usable range to help determine the max range the camera can see within the current ambient light environment.  Some suggestions or settings to maybe look at:  Controls - Invalidation bypass, reduce resolution to 320x240

closed 2021-04-27 by RealSenseSupport