realsenseai/librealsense #8327: rs-software-device example: Not returning depth with breakpoint inside the loop

zc08, 2021-02-08:
| Required Info | |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 |
| Firmware Version | 05.12.06.00 |
| Operating System & Version | Ubuntu 18.04.3 |
| Kernel Version (Linux Only) | 5.4.0-48-generic |
| Platform | PC |
| SDK Version | 2.40.0 |
| Language | C/python |
| Segment | Robot |

### Issue Description
I want to pass some custom rgb-d data to the librealsense pipeline, therefore I'm trying out the [rs-software-device](https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device) example program. 

I've found out that, when the program runs without any breakpoints, both the depth and color data are fine.
However, if a breakpoint was set in the loop, the depth returned holds a NULL pointer, while the color data is fine.

Why does this happen ? Is this related to the syncrhronization mechanism of librealsense ?  Thanks in advance !

MartyG-RealSense, 2021-02-08:
Hi @zc08   A **software_device** case in the link below experienced NULL when a breakpoint was set up.

https://github.com/IntelRealSense/librealsense/issues/6522#issuecomment-639147396

It is a long case.   The solution that worked for the RealSense user in this case to correct their problem was provided here:

https://github.com/IntelRealSense/librealsense/issues/6522#issuecomment-640643730

zc08, 2021-02-08:
Unfortunately,  adding the line: 
`depth_sensor.add_read_only_option(RS2_OPTION_STEREO_BASELINE, depthUnits);`
doesn't work for me.

The curious thing is, the frame that is added earlier gets `NULL`. I.e. If I first add to the color sensor, then depth sensor, then color frame is `NULL`. If I first add to depth sensor, then color sensor, then the depth frame is `NULL`.

MartyG-RealSense, 2021-02-08:
I carefully researched your new information above.  Do you still get a null error if you Stop() the pipeline and **then** activate the breakpoint?

It may also be worth considering whether a **custom processing block** instead of software_device could accomplish oyur goal.

https://github.com/IntelRealSense/librealsense/issues/5847#issuecomment-586718261

closed 2021-02-20 by MartyG-RealSense