realsenseai/librealsense #8329: Image colored blue when ran from terminal but normal on realsense-viewer

sdkaplan, 2021-02-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.07.100  | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 | NVIDIA Jetson Xavier.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  C++                         | 
| Segment			|                            | 

### Issue Description

I am having an odd issue where when I show the realsense image through OpenCV it shows up very blue:
[image]

But when I view the image in the realsense viewer it is normal:
[image]

Here is the relevant code for showing the image:
```

        rs2::frameset data = pipe.wait_for_frames();// Wait for next set of frames
        rs2::frameset aligned_set = align_to.process(data); //align the frames

        rs2::frame depth_filtered = aligned_set.get_depth_frame(); //get aligned depth frame
        rs2::frame color_raw = aligned_set.get_color_frame(); //get aligned color frame

        Mat frame_raw(Size(w, h), CV_8UC3, (void*)color_raw.get_data(), Mat::AUTO_STEP);
        imshow("not tracking", frame_raw);

```
Any ideas on how to get the blue image normal? I think it might be a setting in the camera but I am unsure.

MartyG-RealSense, 2021-02-08:
Hi @sdkaplan   The rendering of color in your image (especially blue instead of pinks and reds) makes me think that it might be an OpenCV **color space** issue, as described in the link below.

https://realpython.com/python-opencv-color-spaces/

[image]

If you are using C++, the RealSense SDK has an OpenCV **Getting Started** example program for displaying the RGB feed from a RealSense camera correctly in an OpenCV windiw.

https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md

MartyG-RealSense, 2021-02-21:
Case closed due to no further comments received.

closed 2021-02-21 by MartyG-RealSense

ychen441, 2023-04-08:
@sdkaplan hey, just switch the cam_config from rgb8 to gbr8, and then your issue will be solved. :)