realsenseai/librealsense #8330: Is it possible to use realsense camera with virtual Android device ?

zc08, 2021-02-09:
| Required Info | |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 |
| Firmware Version | 05.12.06.00 |
| Operating System & Version | Ubuntu 18.04.3 / Android |
| Kernel Version (Linux Only) | 5.4.0-48-generic |
| Platform | PC / Android |
| SDK Version | 2.40.0 |
| Language | C++/Java |
| Segment | Robot |

### Issue Description
Would it be possible to use realsense camera on a virtual Android device ? (E.g. the ones created by android studio) Thanks in advance !

MartyG-RealSense, 2021-02-09:
Hi @zc08   The documentation for the **Android Emulator** in Android Studio states that it does not have virtual hardware for USB devices.  This would likely include a RealSense camera attached to the computer instead of to an Android device.

https://developer.android.com/studio/run/emulator#limitations

zc08, 2021-02-13:
Hi, 
On [this page](https://androidstudio.googleblog.com/2020/08/emulator-30024-canary.html), it seems that the latest android studio emulator does support USB passthrough, so I tried the following command:
```
./emulator -avd Nexus_6P_API_29 -qemu -usb -device usb-host,vendorid=0x8086,productid=0x0b5c
```
where `-avd Nexus_6P_API_29` specifies the name of the android virtual device, and `vendorid` and `productid` are obtained by `lsusb`.

The emulator gives the following error message (showing it 3 times):
```
qemu-system-x86_64: Warning: speed mismatch trying to attach usb device "Intel(R) RealSense(TM) Depth Ca" ( speed) to bus "usb-bus.0", port "1" (full speed)
```

It seems the emulator does find the device, but there's a "speed mismatching" problem. I found a good explanation of the  problem on [this page](https://superuser.com/questions/830890/ipod-touch-5g-wont-recognize-in-qemu-windows-7-vm?newreg=74951413e9cc4f16b89698c7eddca948), saying `usb-bus` corresponds to USB 1.0. Clearly I should use usb-3.0, so I tried the following:
```
./emulator -avd Nexus_6P_API_29 -qemu -usb -device nec-usb-xhci,id=xhci -device usb-host,vendorid=0x8086,productid=0x0b5c
```
specifying `nec-usb-xhci` for usb-3.0.

However I'm getting the error message as:
```
qemu-system-x86_64: Warning: speed mismatch trying to attach usb device "Intel(R) RealSense(TM) Depth Ca" ( speed) to bus "xhci.0", port "1" (full+high+super speed)
```
I noted that, the speed of the camera is displayed as **( speed)**. (I.e. the actual speed, while normally should be one of **(full speed)/(high speed)/(super speed)**, is blank.)
I guess the speed mismatch problem has something to do with this blank speed. Has anyone had similar experiences ? Thanks in advance !

MartyG-RealSense, 2021-02-13:
Thanks very much @zc08 for the update about Android Studio and USB passthrough.

Although I could not find a case that exactly matched your situation, I did find one involving this speed error with a USB HD webcam.  The advice in that case may therefore be transferable to this one involving your RealSense camera.

https://forums.freebsd.org/threads/speed-mismatch-when-passing-usb-device-to-freebsd-running-inside-qemu.77461/

closed 2021-02-25 by MartyG-RealSense