realsenseai/librealsense #8337: On chip calibration gave power state error

sbilgi, 2021-02-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.12.10.00| 
| Operating System & Version |   {Debian strecth}  | 
| Kernel Version (Linux Only)    |  5.9.8                                      | 
| Platform                                 | aarch64  |
| SDK Version                            |  {2.41.0}                          | 
| Language                            |  {c++}                          | 

### Issue Description
Hello all,

When I run the on-chip calibration program was given a power state error (sdk 2.41.0)
But the same code worked properly with SDK 2.34.1.

Here is on chip calibration parameteres: "{\"speed\": 3,\"scan parameter\": 0,\"data sampling\": 0}"
But I tried wtih different parameters. All was same

What is the reason for this kind of error?
What is the main reason for Power state error? Is it relevant to the power supply?

MartyG-RealSense, 2021-02-09:
Hi @sbilgi   Are you experiencing the **Failed to set power state** error please?  This error can indicate a problem with starting a camera stream.

As a starting point in investigating your case, may I ask whether you have you updated your kernel version since the last time that you used on-chip calibration with the On-Chip Calibrator?  I ask this because librealsense officially supports up to **kernel 5.4** and you are using **5.9.8**.

You can test whether it is a kernel related conflict by building librealsense with the **RSUSB backend** installation method.  This method is not dependent on Linux versions or kernel versions and does not require patching.

sbilgi, 2021-02-10:
Hi @MartyG-RealSense, Yes exactly it gave **failed to set power state** error.

No It was the same OS with same kernel. 

I tried with UCV and RSUSB but it gave same error.

MartyG-RealSense, 2021-02-10:
Thanks very much for the information.  Are you performing the on-chip calibration from within the RealSense Viewer or accessing it with script code, please?

sbilgi, 2021-02-10:
Not realsense-viewer, I ran on chip calibration by code (C++).

MartyG-RealSense, 2021-02-10:
And you are using the **C++** code from **Appendix D** of the self-calibration white paper?  The values that you are setting look as though they are quoted from the **C** code (which is a different language to C++) in Appendix B of the paper.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-appendix-d-on-chip-calibration-c-api

sbilgi, 2021-02-10:
Yes It's like Appendix D. Should I run on chip calibration with resolution 256x144?

Here is my sample code

```
int main(){
    
    int m_width = 1280;
    int m_height = 720;
    int m_fps = 15;
    
    
    rs2::config m_config;
    rs2::pipeline m_pipeline;
    rs2::pipeline_profile m_pipelineProfile;
    std::vector<rs2::sensor> m_sensorList;
    rs2::device m_RSdevice;
    
    rs2::context tContext;
    
    m_RSdevice = tContext.query_devices().front();
    m_sensorList = m_RSdevice.query_sensors();
    
    m_config.enable_stream(RS2_STREAM_DEPTH, m_width, m_height, rs2_format::RS2_FORMAT_Z16, m_fps);
    m_config.enable_stream(RS2_STREAM_INFRARED, 1, m_width, m_height, RS2_FORMAT_Y8, m_fps);
    m_pipelineProfile = m_pipeline.start(m_config);
    boost::this_thread::no_interruption_point::sleep_until(boost::chrono::steady_clock::now() + boost::chrono::milliseconds(500));
    bool b_result = true;
    int counter = 0;
    while(b_result){
        try{
            rs2::frameset frameset = m_pipeline.wait_for_frames((1000/ m_fps)+5);
            rs2::video_frame tIRFrameL = frameset.get_infrared_frame(1);
            rs2::depth_frame tDepthFrame = frameset.get_depth_frame();
            std::cout << "Counter: " << counter << std::endl;
            if(counter == 150){
                rs2::auto_calibrated_device tDevice4Calib = m_RSdevice.as<rs2::auto_calibrated_device>();
                float tCalibHealt;
                std::string tJsonCalibParameters = "{\"speed\": 3,\"scan parameter\": 0,\"data sampling\": 0}";
                rs2::calibration_table res = tDevice4Calib.run_on_chip_calibration(tJsonCalibParameters, &tCalibHealt, [&](const float progress) { 
                }, 60000);
            }
            counter++;
        } catch (rs2::error tErr){
            std::cout << "Error in Main: " << tErr.what() << std::endl;
        }
    }
```

MartyG-RealSense, 2021-02-10:
If you are experiencing a **Failed to set power state** area, I suspect that the problem may be in the **query_devices()** section of the code that finds the camera rather than the calibrator code.   The link below provides a C++ example for using query_devices() to get the first detected RealSense device (use the script under the **librealsense2** heading).  This method should work if you only have one camera attached.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-first-realsense-device

sbilgi, 2021-02-10:
Thanks @MartyG-RealSense I think I was used same method with in the [link](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-first-realsense-device). But your sentence gave me inspiration, and I changed getting device method to `m_RSdevice = m_pipelineProfile.get_device();` from `m_RSdevice = tContext.query_devices().front();` Just like in [caliration](https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-appendix-d-on-chip-calibration-c-api) example. Now It's working thanks :) . I think device connection was changed in sdk 2.41. And this is generate problem in my application. Am I right?

Here is working code block:

```
int main(){
    
    int m_width = 1280;
    int m_height = 720;
    int m_fps = 15;
    
    
    rs2::config m_config;
    rs2::pipeline m_pipeline;
    rs2::pipeline_profile m_pipelineProfile;
    std::vector<rs2::sensor> m_sensorList;
    rs2::device m_RSdevice;
    
    rs2::context tContext;
        
    m_config.enable_stream(RS2_STREAM_DEPTH, m_width, m_height, rs2_format::RS2_FORMAT_Z16, m_fps);
    m_config.enable_stream(RS2_STREAM_INFRARED, 1, m_width, m_height, RS2_FORMAT_Y8, m_fps);
    m_pipelineProfile = m_pipeline.start(m_config);
    m_RSdevice = m_pipelineProfile.get_device();
    m_sensorList = m_RSdevice.query_sensors();
    boost::this_thread::no_interruption_point::sleep_until(boost::chrono::steady_clock::now() + boost::chrono::milliseconds(500));
    bool b_result = true;
    int counter = 0;
    while(b_result){
        try{
            rs2::frameset frameset = m_pipeline.wait_for_frames((1000/ m_fps)+5);
            rs2::video_frame tIRFrameL = frameset.get_infrared_frame(1);
            rs2::depth_frame tDepthFrame = frameset.get_depth_frame();
            std::cout << "Counter: " << counter << std::endl;
            if(counter == 150){
                rs2::auto_calibrated_device tDevice4Calib = m_RSdevice.as<rs2::auto_calibrated_device>();
                float tCalibHealt;
                std::string tJsonCalibParameters = "{\"speed\": 3,\"scan parameter\": 0,\"data sampling\": 0}";
                rs2::calibration_table res = tDevice4Calib.run_on_chip_calibration(tJsonCalibParameters, &tCalibHealt, [&](const float progress) {}, 30000);
            }
            counter++;
        } catch (rs2::error tErr){
            std::cout << "Error in Main: " << tErr.what() << std::endl;
        }
    }
}
```

MartyG-RealSense, 2021-02-10:
I cannot think of a specific change that would have caused your particular problem, though there have been numerous SDK releases since 2.34.1 and a very significant amount of updates overall between that version and 2.41.0.  There were also changes in firmware structure from firmware 5.12.8.200 onward that required at least SDK 2.39.0 or newer to be used.

Nevertheless, it is great news that you found a solution - thanks very much for sharing your method!

closed 2021-02-11 by sbilgi