realsenseai/librealsense #8339: realsense-viewer freeze on stopping playback of a BAG file

rkcroc, 2021-02-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 (not attached) | 
| Firmware Version                   | Unknown | 
| Operating System & Version | Win 10 / WSL2 Linux Ubuntu 20.04 LTS |
| Kernel Version (Linux Only)    | 4.19.128-microsoft-standard                                          | 
| Platform                                 | PC |
| SDK Version                            |  2.41.0                          | 
| Language                            | EN-US | 
| Segment			|  other                          | 

### Issue Description

I am using librealsense-2.41.0 on WSL2/Ubuntu 20.04 LTS (kernel 4.19.128-microsoft-standard).
I am loading a BAG file (that I can supply; I cannot upload it here), which then proceeds to play correctly. I am then seeing the viewer freeze when I hit the `Stop` button, as circled in red in the attached screenshot, and then `Start` again. If I instead hit either of the `Pause` buttons circled in green, I do not see the viewer freeze. As I mentioned, when it freezes, it will become completely unresponsive if I restart and stop it again. However, if I unload the file by hitting the X circled in yellow, then it is OK after reloading the bag file.

I get two messages on the console, 
```bash
 08/02 23:20:39,658 WARNING [140441447028480] (playback_device.cpp:198) Playback device does not provide a matcher
 08/02 23:27:19,701 ERROR [140445022592448] (playback_device.cpp:436) Error - timeout waiting for flush, possible deadlock detected
```

Update:
I am now unable to load a BAG file. I get this message:
```bash
08/02 23:57:56,851 ERROR [140075186738944] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info
```

[image]

RealSenseSupport, 2021-02-12:
@rkcroc, could you please check if the issue also happening with our pre-recorded files? 
https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md
I gave "Outdoor scene with D435i..." bag file a try and the 2.41.0 viewer seems to work expected.

RealSenseSupport, 2021-02-18:
Just curious if there's any update on this case and if further help is needed.
Thank you.

maloel, 2021-03-03:
We have this reproduce in C++, and are working on it

rkcroc, 2021-03-03:
@maoel Thanks for the update. Sorry I wasn't able to attempt with your pre-recorded files. If you need a tester for a fix, I am happy to help on a more timely basis.

aangerma, 2021-03-04:
Hello @rkcroc 
The former comment was mistake, we can't reproduce it, I need the bag file for reproduction,
can you please upload it?

rkcroc, 2021-03-04:
@aangerma See attached. I also confirmed that the issue is still happening.
[20200623_111443.bag.zip](https://github.com/IntelRealSense/librealsense/files/6086154/20200623_111443.bag.zip)

closed 2021-03-05 by RealSenseCustomerSupport