realsenseai/librealsense #8341: Segmentation fault when running python 3 with import pyrealsense2

juliahambright, 2021-02-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |T265 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 5.3   | 
| Platform                                 | Raspberry Pi 3b+  |
| SDK Version                            |  librealsense   | 
| Language                            |  python3.6 | 

### Issue Description
Good afternoon! I have installed librealsense on my Raspberry Pi, and I just ran the cmake and other commands described here: https://github.com/IntelRealSense/librealsense/issues/3062#issuecomment-776121080
to get pyrealsense2 for python.

I am using this simple pose code: https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_example.py
and when I run it with python, it works perfectly fine, but when I run it with python3, I get

Segmentation Fault

after a few seconds. I have checked out other posts with similar issues but there does not seem to be a clear solution.

Thanks!

chapa17, 2021-08-24:
Hello, I am also getting a similar error on the code line **pipeline.start(config)**. Did you find any solution to this?

BenSpex, 2021-09-21:
I'm having the same problem on Ubuntu 20.04. 
Has anyone found a solution for this?

henryburon, 2024-02-18:
I got the same error: [ros2run]: Segmentation fault

To fix it, I looked at my video devices with `/dev/video*`

The issue was that there were extra /dev/video* files, and that was confusing librealsense. Remove the files that show up when you have no cameras plugged in. Then, try again and see if your segmentation fault issue goes away. It worked for my raspberry pi, at least.

Unfortunately, however, I'm not experiencing another error:

raceback (most recent call last):
  File "/home/terra/ws/winter_project/install/terraflight_control/lib/terraflight_control/fetch_camera", line 33, in <module>
    sys.exit(load_entry_point('terraflight-control==0.0.0', 'console_scripts', 'fetch_camera')())
  File "/home/terra/ws/winter_project/install/terraflight_control/lib/python3.10/site-packages/terraflight_control/fetch_camera.py", line 70, in fetch_camera_entry
    node = Fetch_Camera()
  File "/home/terra/ws/winter_project/install/terraflight_control/lib/python3.10/site-packages/terraflight_control/fetch_camera.py", line 36, in __init__
    self.pipeline.start(self.config)
RuntimeError: xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
[ros2run]: Process exited with failure 1

I think it has something to do with a driver issue, but I'm not sure. If anyone has any fixes for this, please let me know. I'll keep this updated if I figure out the issue, too.

MartyG-RealSense, 2024-02-18:
Hi @henryburon  When you run your **fetch_camera.py** script, is the camera already in use by another program.  As described at https://github.com/IntelRealSense/librealsense/issues/12377#issuecomment-1801819868 the **Multi-Streaming Model** rules of the librealsense SDK dictate that once a program has claimed access to a stream, another program that is launched afterwards cannot access that stream until the first program has released its claim on the stream.

henryburon, 2024-02-18:
Hi @MartyG-RealSense, no, when I ros2 run the fetch_camera node, the camera is not in use by another program--it's simply plugged into my Raspberry Pi. To fix the issue at the bottom of my previous reply, I have to unplug/re-plug the camera a few times, and then it works. Rebooting also helps.

I am now able to launch the node without errors, and I see initial camera data being published. However, messages soon start being dropped, and I am not able to actually view the images from my laptop on the same network. I think this has something to do with the frequency that I am publishing at and the QoS settings. I'll have to investigate.

MartyG-RealSense, 2024-02-19:
Hi @henryburon  Was the node that you created at https://github.com/IntelRealSense/realsense-ros/issues/3011#issuecomment-1951472270 a solution to the problem described above, please?

henryburon, 2024-02-19:
Yes, it was.

MartyG-RealSense, 2024-02-20:
Okay, thanks very much for the confirmation @henryburon

closed 2024-02-20 by MartyG-RealSense