realsenseai/librealsense #8346: Best setup for long distances?

AJMena, 2021-02-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { l2.41.0.2657 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
We want to measure horizontal distances with the best accuracy. Our target will be between 4 and 6 meters away.
We did some testing, we recorded a wall at 5 meters with different settings, primarily high accuracy. When we observed the results, the wall was too thick, 1 meter for what is a flat surface. We measure some distances, and the average result was not good enough, for example 3.6 meters when the real value was 3.43 meters. We need better accuracy.
We want to know what the best setup for long distances is, lighting, camera preset, post processing filters, ...

MartyG-RealSense, 2021-02-10:
Hi @AJMena   As a starting point in investigating your case, I suggest checking the status of a post-processing filter in the RealSense Viewer called the **Threshold Filter**.

[image]

If it is enabled (indicated by a blue icon beside it) then by default it limits the depth data that is rendered to a 4 meter maximum range.  If the icon beside the Threshold Filter is blue then I recommend left-clicking on it to turn it red, indicating that the filter is disabled.  The D455 will then be able to render depth data for the full distance that it is able to detect depth information.

AJMena, 2021-02-10:
Hello MartyG.
The filter is disabled.

MartyG-RealSense, 2021-02-10:
Would it be possible please to provide images of the location that the camera is being used in, including RGB color?  The RGB image especially can be very useful in diagnosing possible causes of accuracy issues.

AJMena, 2021-02-10:
Yes, of course.
I uploaded the recordings to this drive folder:
https://drive.google.com/drive/folders/1ruAS-mI3SZn27L5JV4d8KV91yULoRoy8?usp=sharing
The recordings were doing in 2 different positions.
For example, you can measure the distance between the end of the line with the "6" above and the beginning of the line with the "luze" above. That must measure 3.43 meters. But in some recordings, you get more, like 3.60 meters.

MartyG-RealSense, 2021-02-10:
Thanks very much for the set of bag recordings.  By checking the settings in the bag, I was able to run depth tests that replicated your results.

In the bags, very high exposure values such as **33000** are being used.  The default depth exposure value is **8500**.  As exposure was increased towards the range that you are using the background disappeared, leaving a small amount of depth detail in the short-range foreground like in your bags.

I would therefore recommend re-testing with the default **8500** exposure value if your project requires using manual exposure instead of auto-exposure.

AJMena, 2021-02-11:
We checked the files, and the auto exposure was enabled. So, the exposure value 33000 is ignored. We did some testing with the value 8500 and manual exposure. Sometimes the result is similar, others, like with high accuracy, the result is worse than with 33000, we get less point.
Is there any other tip we can try out? 
Thank you for your help.

MartyG-RealSense, 2021-02-11:
Using the **High Accuracy** can indeed result in less depth detail as coordinates that have a lower confidence in their accuracy get excluded.  A preset that provides a balance between accuracy and the amount of detail ('fill rate') is **Medium Density**.

Increasing the value of the **Laser Power** setting from its default of '150' (for example, to the maximum '360') can fill in more detail on the depth image by making it less sparse.

For the D455 model the optimal depth accuracy resolution will be **848x480**.  So you may be losing some accuracy by using 640x480.  I can appreciate why using 848x480 at 30 FPS could be a problem if you need to use a limited-capability USB 2.1 connection instead of a USB 3 connection that provides access to the full capabilities of the camera.

In all of the bags though, the key factor in the short depth range seems to be that the **Threshold Filter** was enabled in the **Post-Processing** options when the recording was made, limiting the camera's observable depth range to 4 m.  Disabling the Threshold Filter when playing the bag causes the far wall that is 5 meters away to be rendered, as shown in the side by side montage I created below of the depth with the threshold filter disabled (left) and enabled (right).

[image]

MartyG-RealSense, 2021-02-18:
Hi @AJMena  Do you require further assistance with this case, please?  Thanks!

AJMena, 2021-02-19:
No, thank you.

closed 2021-02-19 by AJMena

MartyG-RealSense, 2021-02-19:
Thanks very much @AJMena for your updates!