realsenseai/librealsense #8349: Unable to configure python build location, which is necessary when using pyenv and virtual environments

zrsmithson, 2021-02-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | NA  | 
| Firmware Version                   | NA | 
| Operating System & Version |   Linux (Ubuntu 18 on Nvidia Jetson NX)  | 
| Kernel Version (Linux Only)    |  4.9.201-tegra                                         | 
| Platform                                 | NVIDIA Jetson NX  |
| SDK Version                            |  2.4.10                         | 
| Language                            |  python, really cmake                         | 
| Segment			|  NA                          | 

CMake configuration does not work with `pyenv`. The build process detects the python location as the active pyenv version, but cmake finds the system library path. This results in a similar output when configuring CMake, which of course fails due to the version mismatch.

```
-- Found PythonInterp: /home/user/.pyenv/shims/python (found version "3.8.6")
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.6m.so (found version "3.6.9")
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.6m.so
```

After searching cmake documentation, I was able to set these variable explicitly, though it is a bit different than opencv building which has more documentation for this sort of thing. The following code blocks show the opencv config and what is needed for librealsense.

Even though I was finally able to solve it, it would be nice if this was documented on the [Build Configuration](https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration) page.

```
PYTHON3_EXECUTABLE:FILEPATH = $(which python3)
PYTHON3_INCLUDE_DIR = $(python3 -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())") 
PYTHON3_LIBRARY = $(python3 -c "from distutils.sysconfig import get_config_var;from os.path import dirname,join ; print(join(dirname(get_config_var('LIBPC')),get_config_var('LDLIBRARY')))")
PYTHON3_NUMPY_INCLUDE_DIRS = $(python3 -c "import numpy; print(numpy.get_include())")
PYTHON3_PACKAGES_PATH = $(python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())")
OPENCV_PYTHON3_INSTALL_PATH = $(python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())")
```

```
BUILD_PYTHON_BINDINGS = ON
PYTHON_EXECUTABLE:FILEPATH = $(which python3)
PYTHON_INCLUDE_DIR = $(python3 -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())") 
PYTHON_LIBRARY = $(python3 -c "from distutils.sysconfig import get_config_var;from os.path import dirname,join ; print(join(dirname(get_config_var('LIBPC')),get_config_var('LDLIBRARY')))")
PYTHON_INSTALL_DIR = $(python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())")
```

There are other hardcoded values like `set(LRS_TARGET  realsense2)` that make it difficult to juggle multiple configurations, but that is beyond the scope of this problem and likely a bigger problem to solve.

Nir-Az, 2021-02-11:
Hi @zrsmithson ,
When I set the "PYTHON_EXECUTABLE" as mentioned at the python wrapper section readme file the build works as expected:

> "Note:_ To force compilation with a specific version on a system with both Python 2 and Python 3 installed, add the following flag to CMake command: -DPYTHON_EXECUTABLE=[full path to the exact python _executable]"

| PYTHON_EXECUTABLE set   | Cmake output |
| -------------------------------------------------------- | ------------------------------------------------------------ |
| PYTHON_EXECUTABLE = C:/Python27/python.exe   | Found PythonInterp: C:/Python27/python.exe (found version "2.7.9") <br/>Found PythonLibs: C:/Python27/libs/python27.lib |
| PYTHON_EXECUTABLE = C:/Program Files/Python37/python.exe | Found PythonInterp: C:/Program Files/Python37/python.exe (found version "3.7.7") Found PythonLibs: C:/Program Files/Python37/libs/python37.lib |

What is the issue you see and what changes did you make to fix it?

MartyG-RealSense, 2021-02-11:
Hi @Nir-Az   Thanks very much for helping @zrsmithson with this case!

zrsmithson, 2021-02-11:
It's working correctly with the right variables set, so updating the documentation for the cmake variables, or putting an entry in the FAQ that specifically mentions pyenv would likely be all that's needed to solve this for others. It might also be worth treating `Python_x` variables the same as `Python3_x` because that's how opencv does it. I can put together a PR for it, but it will be a few days.

I think this only applies when someone has multiple python 3 versions installed, or maybe when it's installed in nonstandard locations. When the executable is provided, cmake doesn't ensure the libs, includes, and site packages actually match. It can be explicitly found using the python executable provider, but probably not necessary to be done in the cmake files.

Really, I tried to solve it with no success, and wrote this issue; then dug through cmake documentation and the realsense cmake files and finally solved it. So hopefully making someone else's life easier in the future.

closed 2021-02-25 by MartyG-RealSense