realsenseai/librealsense #8352: Differences between Realsense viewer's 3D and the corresponding ply

AJMena, 2021-02-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.41.0.2657 }                    | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
In the Realsense viewer I look at the 3D view of the recording of a flat wall with some paintings, all seems flat and the lines are straight. Then I exported a ply of that file. I look at the ply and all seems distorted. I need to look things from the side to see the lines straight. Like I should see the things from where the camera was. If I export the coordinates after using pc.calculate() I get a result similar to the ply.
I can export the same result as the viewer, but I do not think that is correct, feels like cheating the result.
So, what is correct?

[image]
[image]
[image]
[image]
[image]
[image]
The second view is the front view. The third view is somehow from the side to see the "LUZE" line straight.

MartyG-RealSense, 2021-02-11:
Hi @AJMena  The first point to bear in mind when comparing results between the RealSense Viewer and data generated from a program script that you have developed yourself is that the RealSense Viewer applies a range of post-processing filters and depth coloring settings by default, whilst self-created scripts apply no filters by default.  In your own applications, filters have to be deliberately programmed in.  This can cause Viewer images and images produced by scripts to look different.

Also, a ply that is exported from the RealSense SDK and imported into a non-RealSense program to view it may appear distorted if the software that is being used to view the ply is not set to the same **depth scale** that the ply was created at (for example, if the data was recorded in the RealSense Viewer at **0.0001** depth scale but the viewing program is using **0.001** scale).  A ply exported from a program script should use the default **0.001** depth scale unless the script is deliberately programmed to set a different scale value.

MartyG-RealSense, 2021-02-18:
Hi @AJMena  Do you require further assistance with this case, please?  Thanks!

AJMena, 2021-02-19:
No, thank you.

closed 2021-02-19 by AJMena