realsenseai/librealsense #8353: Switch Between Stream Profiles

maltetoelle, 2021-02-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    | 5.8.0-41-generic                                    | 
| Platform                                 | PC |
| SDK Version                            |  2.41.0                   | 
| Language                            |  python                    | 
| Segment			|  Robot                       | 

### Issue Description

I am trying to start a stream with two stream profiles, one with a high and one with a low resolution. I expected that in my callback function both resolution will come in asynchronously. But I always only recieve the resolution, which is the first one in the list in my sensor.open() call. How can I get both resolutions?

Thank you very much in advance!

`class RealSenseStream:

    def __init__(self):

        ctx = rs.context()

        self.depth_sensor, self.color_sensor = ctx.sensors
        assert self.depth_sensor.is_depth_sensor()
        assert self.color_sensor.is_color_sensor()

        high_res_profile = self.color_sensor.profiles[6].as_video_stream_profile()
        normal_res_profile = self.color_sensor.profiles[24].as_video_stream_profile()

        self.color_sensor.open([normal_res_profile, high_res_profile])

        self.frame = None
        self.color_sensor.start(lambda f: self.update_frame(f))

    def update_frame(self, f):
        self.frame = np.asanyarray(f.get_data())`

ev-mp, 2021-02-11:
@maltetoelle,
You can configure each stream for one configuration (res/fps/format) per streaming session. This is fundamental with UVC.
To switch resolutions you need to stop and reconfigure the sensor.

maltetoelle, 2021-02-11:
Thank you very much for your answer! I solve my issue like so now, which is kind of slow

`
    def get_high_res_image(self):
        self.color_sensor.stop()
        self.color_sensor.close()
        self.color_sensor.open(self.high_res_profile)
        self.color_sensor.start(self.syncer)
        frames = self.syncer.wait_for_frames()
        color_frame = np.asanyarray(frames.get_color_frame().get_data())
        self.color_sensor.stop()
        self.color_sensor.close()
        self.color_sensor.open(self.normal_res_profile)
        self.color_sensor.start(self.syncer)
        return color_frame
`

Must I close the sensor or can I switch after I called stop() between active streams that can be received via sensor.get_active_streams()?

ev-mp, 2021-02-11:
Yes, you must call sensor.close() as the handshake with device and stream profile selection occurs in sensor.open().
Actually, the data from device starts streaming on sensor.open().
Calling sensor.start() just invokes polling the frames from kernel/driver to SDK/user.

maltetoelle, 2021-02-11:
alright, thank you for your answer! That sadly takes quite long untill one can go back to "normal" streaming. What are the active streams then for?

ev-mp, 2021-02-11:
The active streams is a subset of profiles the camera is streaming at a given moment (one per stream)

maltetoelle, 2021-02-11:
ok, thank you!

closed 2021-02-11 by maltetoelle

ev-mp, 2021-02-12:
@maltetoelle , you can get the desired effect of multiple resolutions per a stream using [decimation](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#decimation-filter) post-processing filter . See https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb tutorial.