realsenseai/librealsense #8354: IR stream interference

rupak-d, 2021-02-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.12.06.00 | 
| SDK Version                           | 2.33 | 

### Issue Description
Is it possible for the infrared stream of the D435 cameras to interfere with elevator door closing sensors?

MartyG-RealSense, 2021-02-12:
Hi @rupak-d   RealSense 400 Series cameras do not interfere with each other, though they can interfere with non-RealSense infrared devices.

MartyG-RealSense, 2021-02-25:
Case closed due to no further comments received.

closed 2021-02-25 by MartyG-RealSense

rupak-d, 2021-02-25:
Is there any way of distinguishing if there are multiple IR sources?

MartyG-RealSense, 2021-02-25:
Apparently there is a phenomenon that can be performed with a smartphone where if you switch to **front camera** view and film a scene, sources that are emitting infrared (like the red flash on a remote control when pushing buttons) will show up as a visible light in the digital camera image  on the phone screen.

The reason why front camera mode should be used is that there may be an IR filter on the rear camera that prevents IR light sources from being visible on the camera image.
.
https://www.carolina.com/knowledge/2020/02/20/make-the-invisible-visible

rupak-d, 2021-03-01:
Sorry I meant is there a way for the D435 cameras to distinguish between it's own IR vs one from a different source?

MartyG-RealSense, 2021-03-01:
The 400 Series cameras can see all wavelengths from the visible to near IR.  So as far as I know, there is no built-in function for identifying which IR wavelength belongs to which source.  It should be able to **see** the wavelength though.

https://github.com/IntelRealSense/librealsense/issues/1070#issuecomment-366405803

The projector on the 400 Series cameras uses the **850 nm** wavelength.

rupak-d, 2021-03-01:
Thanks for the info!
We've seen two situations where the camera is stopping elevator doors from closing when pointed at it. Any ideas on how to solve this would be welcome.

MartyG-RealSense, 2021-03-01:
The bottom of Section 4.3 of Intel's white-paper document on physical **optical filters** states how using **notch filters** can block wavelengths and so enable devices to co-exist with each other.  Perhaps such a filter could be applied over the top of the elevator door sensors so that they ignore the wavelengths of the 400 Series camera.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters

To quote the section:

"The use of a notch filter (essentially the inverse of a bandpass filter) would be useful in scenarios where there are specific wavelength bands that may interfere with or reduce the depth quality.  For example, if another device is using a laser of a specific wavelength to illuminate or scan a scene, then a notch filter can be used to block only that wavelength.  This approach is sometimes used to let devices that operate at 850nm and 940nm co-exist without affecting each other at all".