realsenseai/librealsense #8355: C++ max distance

oscarthorn, 2021-02-11:
Hi!

I'm using the D455 with the c++ API. What is the default max distance setting (is there one) and how can I change it?

Cheers!

MartyG-RealSense, 2021-02-12:
Hi @oscarthorn   If a **threshold filter** to set a custom minimum and maximum depth sensing distance is not being applied (which it would not be by default if you are creating your own C++ application) then the camera would use its full depth sensing range. 

The links below provides example C++ code for defining a threshold filter.

https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp#L74

https://stackoverflow.com/questions/59054413/intel-realsense-depth-camera-d435i-noises-and-mounds

oscarthorn, 2021-02-12:
Thanks! Great!

closed 2021-02-12 by oscarthorn