realsenseai/librealsense #8356: Jetson python rs.config() returns Illegal instruction (core dumped)

Lostosaurus, 2021-02-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> 2.41.0) | 
| Operating System & Version |   {L4T 32.5.0 [ JetPack UNKNOWN ] Ubuntu 18.04.5 LTS | 
| Kernel Version (Linux Only)    |  (4.9.201-tegra)                                         | 
| Platform                                 | Jetson Xavier NX  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| opencv-python                            |  {4.5.1.48}                          | 
                

### Issue Description
Attempting to run the example code ''opencv_viewer_example.py'' fails with the error
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py
Illegal instruction (core dumped)

it seems this line is causing errors for me.
config = rs.config()

I am able to run realsense-viewer and run the example code ''python-tutorial-1-depth.py ''

the cmake i used was as follows.

cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true -DFORCE_LIBUVC=ON

I also checked these two places, and was wondering if i had old installations lingering around.

ls -la /usr/lib/x86_64-linux-gnu | grep realsense
returns nothing

ls -la /usr/local/lib | grep realsense
lrwxrwxrwx  1 root root         24  2月 12 01:58 librealsense2-gl.so -> librealsense2-gl.so.2.41
lrwxrwxrwx  1 root root         26  2月 12 01:58 librealsense2-gl.so.2.41 -> librealsense2-gl.so.2.41.0
-rw-r--r--  1 root root   23512208  2月 12 01:20 librealsense2-gl.so.2.41.0
lrwxrwxrwx  1 root root         21  2月 12 01:58 librealsense2.so -> librealsense2.so.2.41
lrwxrwxrwx  1 root root         23  2月 12 01:58 librealsense2.so.2.41 -> librealsense2.so.2.41.0
-rw-r--r--  1 root root  208776200  2月 12 01:20 librealsense2.so.2.41.0
-rw-r--r--  1 root root   13691180  2月 11 11:21 librealsense-file.a

RealSenseSupport, 2021-02-12:
Hi @Lostosaurus 

There is work being done on the Python examples to make them compatible with L515.  Previous releases of the python examples were by default configured for resolutions and device configurations for the D400 cameras.  Some of these defaults are not compatible with the L515, some of the resolutions aren't supported on the streams for the L515.  With that being said, you can have a look at the development branch where some of the examples have been updated and give them a try:  https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python/examples

Lostosaurus, 2021-02-13:
Thank you for your response. @RealSenseSupport 

I have tried the scripts noted as being updated for L515. python-tutorial-1-depth.py works without trouble, though I did need to change import pyrealsense2 to import pyrealsense2.pyrealsense2.

opencv_viewer_example.py returns : Illegal instruction (core dumped)
https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/opencv_viewer_example.py

the frame_queue_example.py returned: 
Slow callback
Segmentation fault (core dumped)

Lostosaurus, 2021-02-13:
Hello, I found the issue after comparing the depth code and the opencv code. It appears that the numpy version 1.19.5 causes issues with OPENBLAS_CORETYPE and causes everything to crash. I apologize for posting here when the issue was with numpy instead. downgrading to numpy 1.19.4 solved my issue.

https://github.com/numpy/numpy/issues/18131

closed 2021-02-15 by Lostosaurus