realsenseai/librealsense #8357: Intel RealSense LIDAR L515 not being detected by windows after firmware update via Intel.RealSense.Viewer.exe (2.41.0.2657)

tkwane, 2021-02-12:
Hi everyone please assist,

Recently acquired an L515 LIDAR to aid with computer vision research.

First-time use connected it to the computer and the drivers were automatically installed and the device was identified by windows successfully. I then proceed to download the Intel.RealSense.Viewer.exe just to verify the lidar is working properly and I can see the depth information. Upon the first launch of the above-named program, there was a prompt to update the firmware in order to proceed. I clicked update, the updating firmware process was showed on the top right-hand corner of the screen, it seemed to progress through the update successfully. Consequently, there was an error message saying the device took longer to reconnect to the software than expected or something similar to that effect. I allowed it to stand for a few minutes but no response from the camera. I have since then disconnected and reconnected the camera to the computer too many times to compatible USB3 ports. I have even installed the SDK and retried connecting but the device is not being identified by windows. I later connected the device to the next computer and repeat the process of installing the SDK but all attempts proved futile. 

Below is attached the last driver events from the system I was using when the device failed to be detected by windows after the update was done by Intel.RealSense.Viewer.exe (2.41.0.2657)?

I have so many questions.

[image]
[image]
[image]
[image]
[image]
[image]
1. How can official intel software prompt to update on first use and the device fail immediately?

2. Is the device faulty?

3. I have been consumed in searching the internet and the community for similar reported issues. I have seen a lot of similar issues but none of the recommendations there have been helpful.

4. I Have tried using the update firmware instructions provided on GITHUB. However, the console provided in Intel.RealSense.Viewer.exe does not identify any commands. I tried navigating directly to the installation directory to load the update firmware tool (Intel.RealSense.FW.Update.exe) and it starts for a second display some writing on the screen and closed much too quickly for you to see or interact with the writing on the screen.

4. I have read about DFU mode, But since then have not been able to find any useful info to assist on that topic. 

5. The device was purchased on amazon by the Electrical Engineering Department at the university. Should I immediately report the issues as a faulty device or is this a really simple resolvable issue?
6.

RealSenseSupport, 2021-02-12:
Hi @tkwane 

Sorry that you're running into these types of issues with the L515 camera.  I have a couple questions that may help us understand what could be going on and determine if this is a HW issue or SW issue.

- Can you tell if the camera is recognized by Windows as a USB2 device or USB3 device?  USBView can be used for this.
- The camera is a standard Winusb device, so no other drivers are necessary...can you run rs-enumerate-devices and see if the camera is being seen by this tool? https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

tkwane, 2021-02-12:
Just installed the USBview. However the device is not being detected by USBview on any of the compatible USB ports on my system. The screenshots shown on initial post was display via enabling hidden devices to be shown.  I am currently attempting the rs-enumerate-devices. 

Please clarify if the commands are to be run from windows PowerShell or natively inside the IntelRealSense debuging console provided in Intel.RealSense.Viewer.exe (2.41.0.2657) **Really need some clarification on this thank you**

tkwane, 2021-02-13:
I ran the command in Intel.RealSense.Viewer.exe (2.41.0.2657) debug console but again it says unrecognized command.
I tried navigating to the windows folder and launching the rs-enumerate-devices.exe and its not starting. 

I navigate to PowerShell and called the "rs-enumerate-devices.exe" and additively used the flags. The sole response from the application was that the device is not being detected despite being connected to the computer on compatible USB3 port.

Please see screenshots below.

[image]
[image]

RealSenseSupport, 2021-02-19:
Thanks @tkwane for the debug information you provided here.

There seems to be something wrong with the HW if the device isn't being detected by the system.  I suggest investigating RMA process for the camera.  Please work with the store in which you purchased device from whether that is the intelrealsense.com webstore or distributor.

RealSenseSupport, 2021-02-25:
Hi @tkwane 

If you've started the RMA process, can you outline your camera serial number that will be sent through the RMA process?

plsun, 2021-03-09:
I have very similar problems. 
1.	My L515 connected as USB3.2 with Intel RealSense Viewer v.2.41.0 only in first two days I purchased. Then it turned to USB2.1 with low spatial resolution and never returned. I tried it on 3 different Win10 laptops. All got the same results.
2.	Afterward, I tried many ways searched from GitHub to update the firmware. I used rs-fw-update –f Signed_Image_UVC_1_5_4_1.bin (the image file is in the folder). Not work.
3.	Finally, I used rs-fw-update –r. Thereafter, the driver cannot be detected by Windows anymore. All Intel provided tools cannot detect the device (including rs-enumerate-devices).
4.	I am going to ask the dealer to repair it. I have D415 and D435, they are working well.

tkwane, 2021-03-09:
Hi @plsun ,

I had resolved this issue. 

You will need the computer in which you first connected the device and the drivers were installed correctly.

Connect the device back to that computer in compatible USB 3 port. 
Then ensure your have the IntelRealSense-2.41.0.2657 SDK installed.
Then navigate to the installation directory where you can find the rs-enunerate**.exe  and open PowerShell in this directory. 

Next download the most up to date firmware. Extract the firmware in the same folder you opened PowerShell in above. 

Then execute the following instructions as seen in the image below.

Your welcome.

[image]

RealSenseSupport, 2021-03-24:
Thanks @tkwane for that information

@plsun were you able to recover?

tkwane, 2021-03-25:
You are most welcome.

plsun, 2021-03-25:
Sorry, the method recommended by @tkwane did not solve my problems. Finally, I returned it for repair two weeks ago. Anyway, thank very much for your kindly help.

RealSenseSupport, 2021-03-29:
Thanks for the update

closed 2021-03-29 by RealSenseSupport