realsenseai/librealsense #8363: Announcement: librealsense SDK version 2.42.0 now available

MartyG-RealSense, 2021-02-14:
Hi everyone,

Version **2.42.0** of the librealsense SDK is now available from the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0

**API changes**
https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2420

**Release Notes**
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2420

akifh, 2021-02-15:
@MartyG-RealSense , thanks for the announcement. There seems to be something missing. On release page (https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0) it recommends firmware 5.12.11.0, however, on firmware page (https://dev.intelrealsense.com/docs/firmware-releases) the latest one is 5.12.10.0.

[image]

[image]

Is this a typo, or should we wait for 5.12.11.0 a little more?
Thanks,
Akif

MartyG-RealSense, 2021-02-15:
Hi @akifh   New firmwares are bundled into the SDK that they are recommended for.  So you can update the firmware in the RealSense Viewer program by going to the **More** option near the top of the Viewer's options side-panel and selecting from its menu the option **Install recommended firmware**.  This should automatically update your camera to the new firmware without needing a firmware file.

The standalone bin file for a new firmware usually appears on the firmware download page within a week or so of its initial availability at the release of a new SDK version, so I recommend checking the firmware page occasionally to see when the latest firmware appears on it.

[image]

akifh, 2021-02-15:
@MartyG-RealSense , thanks for the info. I will deploy the new firmware to a remote, headless system via rs-fw-update tool. Therefore, viewer etc. are not available, I need bin file. Then, we will wait for it to become available. Maybe, for upcoming releases, recommendations and firmware download page can be more synchronized.

Thanks,
Akif

MartyG-RealSense, 2021-02-15:
My expectation would be that if you have the previous firmware **5.12.10.0** currently installed then it should still work for the 2.42.0 release.

akifh, 2021-02-15:
@MartyG-RealSense , thanks, yes I think so. But this is something irrelevant with synchronization of both pages. As soon as a firmware is recommended officially, one should be able to use it with official rs-fw-update tool, without needing GUI tools etc.

Thanks,
Akif

MartyG-RealSense, 2021-02-15:
@akifh  I do understand your concern about release synchronization.  Considering the logistics involved, Intel are agile in responding to a new librealsense release with updates for supporting components (websites, wrappers, etc).

ev-mp, 2021-02-15:
@akifh hello, the distribution method involves passing external hosting services, and, as pointed by @MartyG-RealSense , it may take certain time for all the release content to propagate.
Specifically for the new FW - running SDK's CMake locally will download the new image bin files, and those can be used immediately.

akifh, 2021-02-15:
@MartyG-RealSense day-one updates may not be possible due to external services, yes, but "a week or so" was a bit long for me so I asked for better synchronization. Though, @ev-mp 's suggestion is sufficient for me at this point. 

Thanks for all the replies,
Best,
Akif

sam598, 2021-02-15:
@agrunnet @dorodnic @MartyG-RealSense Any update on external color synchronization for the D455 cameras? https://github.com/IntelRealSense/librealsense/issues/7502

smeng9, 2021-02-15:
Still no official release for latest python version? Disappointed

MartyG-RealSense, 2021-02-15:
@smeng9   2.42.0 includes the updates to **pybind11** discussed in https://github.com/IntelRealSense/librealsense/issues/8229#issuecomment-767348365 

**Integrate pybind11 V2.6.1 + replace pybind11 files with clone action
Clone pybind11 instead of keeping pybind files inside librealsense2 repo
Update pybind version 2.2.1 -> 2.6.1**

smeng9, 2021-02-15:
Understand, but the pypi release will be much easier and helpful because we no longer need to build from source when deploying on multiple computers

MartyG-RealSense, 2021-02-15:
@sam598  I have no details to provide about external color synchronization for the D455 cameras at the time of writing this.

closed 2021-02-16 by MartyG-RealSense

MartyG-RealSense, 2021-02-17:
Hi @akifh  Firmware **5.12.11.0** is now listed on the firmware download page.

https://dev.intelrealsense.com/docs/firmware-releases