realsenseai/librealsense #8364: Sunlight / outdoor operation

ShettyHarapanahalli, 2021-02-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   |  
| Operating System & Version |  Ubuntu 20| 
| Kernel Version (Linux Only)    |                              | 
| Platform                                 | Raspberry Pi/ NVIDIA Jetson  |
| SDK Version                            |                   | 
| Language                            |  /opencv/python                          | 
| Segment			|  Robot                          | 

What is the depth error expected when we have to operate camera in a sunlight ex- window in the operation area, sometimes the camera may have to operate in a view where sun appears into the camera visoin. 

Also is there any updated camera with an IP 6x rating coming soon.

MartyG-RealSense, 2021-02-15:
Hi @ShettyHarapanahalli   RealSense 400 Series cameras work excellently in strong natural light, as described in the section of Intel's camera tuning paper linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-sunlight-but-avoid-glare

As this documentation mentions though, there may be negative consequences for the image if the sunlight is causing glare to reflect off a window pane.  The impact of such glare can be significantly reduced by applying a physical optical filter called a **linear polarization filter** over the top of the lenses on the outside of the camera.  Section 4.4 of Intel's white-paper document about optical filters provides more information about this.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters

The image below, extracted from Section 4.4, illustrates the difference that a linear polarization filter can make to negating glare.

[image]

Intel does not discuss possible future camera models.  You are free though to create a dust-proof custom housing for the camera using methods such as 3D printing though and use your own choice of transparent cover material to place over the camera lenses.  The link below provides useful research leads for optical filters and custom cover materials.

https://github.com/IntelRealSense/realsense-ros/issues/1232#issuecomment-640483937

ShettyHarapanahalli, 2021-02-16:
Thanks a ton for sharing this information, I would just wait for IP6X rated stereo camera with an inbuild ai accelerator and POE to come soon to the market. Will wait for the same.    :)

closed 2021-02-16 by ShettyHarapanahalli