realsenseai/librealsense #8367: Pi4 and Python

epicsurf5200, 2021-02-15:
Hi I'm a fairly new programer and am still learning the roots. I am trying to get my realsense d435i to run on my Raspberry pi 4. Ive tried many different operating systems and can't seem to get the viewer to work... The only success ive had is when I used my Pi as a server for the camera and was able to connect to realsense viewer on my desktop. 
I have tried the sample python codes producing the error "unknown directory"
I have also tried the CPP examples and produce errors "no such file or directory" when trying to build.
I assume the library is not installed correctly. 
Is Rasbian a good OS for the realsense?
Is there a tutorial you would recommend following to install the directory?

Any help troubleshooting will be greatly appreciated.

MartyG-RealSense, 2021-02-15:
Hi @epicsurf5200   An installation guide for Pi 4 that is popular with RealSense users is the **Acrobotic** guide:

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

Alternatiuvely, a highly technical guide for Pi 4 and Raspbian Buster that includes additional instructions for ROS installation and guidance for what to do when errors in the installation occur is **eleccelerator**:

https://github.com/IntelRealSense/realsense-ros/issues/1375#issuecomment-691482673

epicsurf5200, 2021-02-16:
Thank you! I am working through the tutorial and im getting an error when building the python C++ implementation.
When installing protobuf, everything seems to work fine until I run the command "python3 setup.py build --cpp_implementation". 

pi@raspberrypi:~/protobuf/python $ python3 setup.py build --cpp_implementation 
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    from distutils import util
ImportError: cannot import name 'util' from 'distutils' (/usr/lib/python3.7/distutils/__init__.py)

MartyG-RealSense, 2021-02-16:
A couple of RealSense users have found that Protobuf issues with this guide and Intel's own Raspbian one are resolved if Protobuf 3.14 is used.  It may be worth trying that first before exploring other options regarding this error.

https://github.com/IntelRealSense/librealsense/issues/7927

epicsurf5200, 2021-02-17:
Im still getting the same error. I used the command <git clone --depth=1 -b v3.14.0 https://github.com/google/protobuf.git> to get the specific version. Is this the right way to go about it? I am using the latest version of Rasbian buster without the desktop. Should I try an older OS?

MartyG-RealSense, 2021-02-17:
I have not seen this particular error **ImportError: cannot import name 'util' from 'distutils'** before.  You could check whether **distutils** is installed on your Python environment by using the installation command below to install distutils.

**sudo apt install python3-distutils**

epicsurf5200, 2021-02-17:
I installed protobuf using the guide on github and seemed to solve the error I was having. However, I have a new error that occurs on the same step.

pi@raspberrypi:~/protobuf/python $ python3 setup.py build --cpp_implementation
Traceback (most recent call last):
  File "setup.py", line 12, in <module>
    from setuptools import setup, Extension, find_packages
ModuleNotFoundError: No module named 'setuptools'

Could this be caused by an issue with the python installation?

epicsurf5200, 2021-02-17:
I just checked the version of python I am using and it is 3.7.3

MartyG-RealSense, 2021-02-17:
Another RealSense user had this error and overcame it by using this instruction:

**sudo pip3 install setuptools**

https://github.com/conan-io/conan/issues/7092#issuecomment-634047096

epicsurf5200, 2021-02-17:
Thank you, I really appreciate your help. I actually just figured out that error. but now im getting another error..

google/protobuf/pyext/message.cc:502:12: warning: ‘int google::protobuf::python::VisitCompositeField(const google::protobuf::FieldDescriptor*, PyObject*, Visitor) [with Visitor = google::protobuf::python::cmessage::SetOwnerVisitor]’ used but never defined
google/protobuf/pyext/message.cc:502:12: warning: ‘int google::protobuf::python::VisitCompositeField(const google::protobuf::FieldDescriptor*, PyObject*, Visitor) [with Visitor = google::protobuf::python::cmessage::FixupMessageReference]’ used but never defined
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

not really sure what is causing this..

MartyG-RealSense, 2021-02-17:
My experience with programs in general is that if the installation process is arguing with you at every step then something fundamental has gone wrong somewhere in the process.  The temptation is try to force through and fix each individual step, but it can be easier to backtrack and find where the process might have originally gone wrong.

It is also worth mentioning, if you have not found it already, that Intel provide a pre-made librealsense image for Pi 4 that can be flashed onto a SD card.  The librealsense version on the image is quite old, but it can be useful to use as a reference for creating your own librealsense installation on Pi.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

epicsurf5200, 2021-02-17:
Ill give that a try. On this prebuilt image are wrappers installed? My end goal is to be able to use OpenCV on a python script for the camera. Will I need to install those or are they included in the build?
Thanks

MartyG-RealSense, 2021-02-17:
The SD card build does not include wrappers.

closed 2021-03-01 by MartyG-RealSense