realsenseai/librealsense #8369: L515: sensor mode (Vga) with RS2_L500_VISUAL_PRESET_CUSTOM is incompatible with the requested profile resolution (Xga)

ruminus, 2021-02-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.01.03 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8.0-43-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.42.0                          | 
| Language                            |  Python                          | 

### Issue Description
Can't start the stream (depth and color) with 1024x768 depth resolution, getting following error:
`sensor mode (Vga) with RS2_L500_VISUAL_PRESET_CUSTOM is incompatible with the requested profile resolution (Xga)`

Streaming with 640x480 works. No custom presets are initialized.

Found an option called `sensor_mode` (`Notify the sensor about the intended streaming mode. Required for preset `) which can have values `0.0-2.0`, corresponding to vga, xga and qvga respectively. The default value is `0.0`. Setting the option to `1.0` does nothing, `get_option` always returns `0.0` unless you bind the sensor object to a variable.

Also the culprit seems to be setting some depth options before the stream:
```
l515_depth_options = {
    'pre_processing_sharpening': 5.0,
    'confidence_threshold': 3.0,
}
```

If we don't set the options, the stream starts without error, but we need them to combat noise in certain conditions, otherwise PC is unusable.

maloel, 2021-02-18:
Hi @ruminus 

The exception you are referring to has since been removed because it was problematic. See PR #8338. I suggest you cherry-pick it if it continues to be a problem as that's the easiest solution.

That said, if you do not take the PR:

`sensor_mode` controls the intended resolution for streaming. However, if the requested profile resolution differs, you can get this exception but only in CUSTOM visual preset. I.e., if you can reset the visual preset to a standard preset, you also will not get this exception. You mat not be able to -- I see that you set options which put you in CUSTOM.

This option `sensor_mode` is a memory-only option (not on the camera) -- so if you destroy the sensor object (or use a new one) after setting the option, then the value is lost.

I hope all that helps!

ruminus, 2021-02-19:
@maloel  thanks! Tested today with cherry-picked commits, everything works.

closed 2021-02-19 by ruminus