realsenseai/librealsense #8373: D455 Not reading depth outside 4m

schleptech, 2021-02-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |             Realsense D455 not reading outside 4m                                 |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | (05.12.10.00) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | Jetson Xavier  |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  Machine Vision                           | 

### Issue Description

Depths not showing outside 4m on realsense viewer. Specifications state that ideal range is 6m, but depths not showing
Also - lots of noise at 4m. How do I reduce noise? (see attached video)
Application is scanning vehicles using realsense to build 3d model supplemented by LiDAR sensors

https://user-images.githubusercontent.com/71199728/108107669-2a59ee80-7055-11eb-978a-70a67daebb23.mov

[image]

sam598, 2021-02-16:
Are you using the high accuracy preset? I know that it reduces the amount of depth data based on confidence, so it may be you are not getting data at a further distance. Do you get data at 6m on the default preset?

The specs say that the depth accuracy is <2% at 4m, which I assume means it could be off by 80mm/3in. I'm honestly not sure if intel is referring to literal accuracy, or precision noise.

Also cars are notoriously difficult to scan. If you are scanning in direct sunlight it is likely that the environment is too bright for the laser projector to be visible, and the camera is relying purely on a stereo image for depth estimation. Are you able to get better car reconstruction at night or in the shade?

MartyG-RealSense, 2021-02-16:
Thanks very much @sam598 for your help to @schleptech 

This 4 meter limitation with D455 is usually related though to a **post-processing** filter in the RealSense Viewer called the **Threshold Filter** being enabled by default when the Viewer is launched.  This filter is set to a default maximum depth of '4' (meters), meaning that depth data outside of its defined minimum and maximum range is excluded from the depth image.  

If the filter - found under the **Post-Processing** section of the **Stereo Module** options - is enabled by default (indicated by a blue icon beside it that means it is On), it can be disabled by left-clicking on the blue icon to toggle it to red (off).  The Viewer will then be able to display the full range of detectable depth information streamed by the D455.

[image]

schleptech, 2021-02-16:
@sam598 send me an email Thomas AT schleptech dot com ... I have a few questions for you.

@MartyG-RealSense Thanks for the information, it looks like I was able to unlock that further data through your instructions!

closed 2021-02-16 by schleptech

xiangtanbailanwang, 2024-10-28:
I ran into the same trouble as you,thanks a lot from all you big shots for helping out!