realsenseai/librealsense #8374: T265 stops unexpectedly on Realsense Viewer 2.42.0 on Raspberry Pi 4 Ubuntu 20.4

MarcoAndrade26, 2021-02-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265| 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |   Ubuntu 20.04.2 LTS && Ubuntu 18 LTS| 
| Kernel Version (Linux Only)    |  5.4.0-1028-raspi | 
| Platform                                 | Raspberry Pi 4 4Gb  |
| SDK Version                            |  2.42.0                           | 
| Language                            |  C++                         | 
| Segment			|  Robot                          | 

### Issue Description
I am trying to use T265 on Realsense Viewer, but after a while and randomly, it shows me the error "No Frames Recieved!"

See image:
[image]

Trying to solve it, I click on "Hardware reset" and on console It shows me this:
[image]

Then I close realsense viewer, open it again and camera works fine for a while, but then it stops and shows me the same error "No Frames Recieved". I tried with several solutions:
- Connecting camera to a USB 3.0 Powered Hub
- Connecting the camera directly to the Pi to USB port 3.0

zlite, 2021-02-18:
Same problem here

MarcoAndrade26, 2021-02-22:
@RealSenseSupport @RealSense-Customer-Engineering @RealSenseCustomerSupport

PradoCg, 2021-02-22:
I have the same problem, in my case the Camera connects for a few seconds and detects itself. Incredible as the manufacturer has no solution for this.

MarcoAndrade26, 2021-02-28:
I still cannot solve this issue, I am going to buy D435i camera. I do not know why Intel launch a product if it is not going to give support.

JeremyBYU, 2021-03-27:
I am also facing this issue. I had previously had everything working on the RPI4 on version 2.35. However now that I have updated to version 2.39+ the T265 stops working after a few seconds, sometimes a minute. 

I attempted to downgrade to a previous working version. I am now on 2.38.1 but now it cannot recognize the T265. `rs-enumerate-devices` returns nothing. I'm wondering if it is automatically updating firmware on the T265 device. If so, that's a lot different than my previous experiences with the D4XX and L515 cameras, where this a user initiated process. 

Not sure where to go from here.

Twentystudios, 2021-03-28:
I’m having the exact same issue. Hope to see a solution for this soon.

JoseDinis1, 2021-04-09:
Have the same problem https://github.com/IntelRealSense/librealsense/issues/8779, and dont see any replies from Realsense. I dont understand then why they sell this, but isnt supported. Its the second hardware i buy from intel that is problematic.

I tried the RS 2.37, and 2.35 in Raspberry Pi Os (not in ubuntu) and have the same problem.

JoseDinis1, 2021-04-09:
Have the same problem: tried the RS 2.37, and 2.35 in Raspberry Pi Os (not in ubuntu) and realsense still doesnt work properly

csr-kick, 2021-04-09:
#6362
#6707 
#6598

and many more...
https://github.com/IntelRealSense/librealsense/issues/8779#issuecomment-816983561

Bercon, 2021-04-27:
Facing similar issue on Raspberry PI 4 and Ubuntu 20.4.2 LTS using librealsense in C++, latest master branch built from sources. It runs fine for couple seconds and then crashes, often freezing the whole OS.

MarcoAndrade26, 2021-05-31:
Hello! 

I believe it is due to the lack of memory in the Raspberry Pi. I changed to Jetson AGX Xavier and camera works fine

closed 2026-10-23 by MartyG-RealSense