realsenseai/librealsense #8375: Inconsistent exposure setting

fiorano10, 2021-02-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  4 x D435i | 
| Firmware Version                   | 5.12.10.0 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.3.62                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.41 |
| Language                            |  C++                         | 
| Segment			|  {Robot                          | 

### Issue Description
We're running multiple D435is with a 2-stop Neutral Density filter on each infrared camera. It appears when 2 cameras are looking at the same scene, the exposure is vastly different between them when then scene is bright, as the scene gets darker the exposure level equalizes. The graph below shows the 2 right cameras that are brighter with higher exposure and the left which are darker with lower exposure.

[image]
[image]

MartyG-RealSense, 2021-02-17:
Hi @fiorano10  It would be best to focus the discussion on https://github.com/IntelRealSense/librealsense/issues/8129 where your case is already being handled and has the attention of Intel RealSense team members.  We could leave this new post up for a time to see whether other RealSense users report similar symptoms to yours though.

MartyG-RealSense, 2021-03-21:
Hi @fiorano10   Although the related case https://github.com/IntelRealSense/librealsense/issues/8129 was closed, can you confirm whether you still require assistance with this particular issue or whether it can be closed now?  Thanks!

closed 2021-03-27 by MartyG-RealSense