realsenseai/librealsense #8379: Realsense camera Openni2 support in OpenCV

Novruz97, 2021-02-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435i | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Ubuntu 18.04, melodic | 
| Kernel Version (Linux Only)    | 5.4.0.65 - generic                                      | 
| Platform                                 | PC |
| SDK Version                            |  2   | 
| Language                            |  python, opencv, openni2                          | 
| Segment			|  camera compatibility                        | 

### Issue Description
Hello dear developers. I am learning coding with Opencv in Python using the book "Learning OpenCV4 with Python3". In the book it mentions to use Openni2 to work with depth cameras. I have surfaced in the web to find out if Openni2 supports Intel Realsense D435i camera, and it seems like it can. When I proceeded to use VideoCapture(cv2.CAP_REALSENSE), the compiler didn't show any result, neither it showed any errors. 
I searched for any other examples of Openni2 usage with Opencv for D400 series, but haven't found. So the question is, is it possible to use Openni2 in Opencv for D400 series? 

P.s. I know that with librealsense2 I can use the camera in opencv. I just want to know if its possible with Openni2. Thank you!

MartyG-RealSense, 2021-02-17:
Hi @Novruz97  Yes, librealsense has a wrapper for Openni2 that allows a 400 Series camera to be used with it.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2

In regard to using Openni2 with OpenCV, the RealSense community member **UnaNancyOwen** (known in the community for the bag-file image extractor program **rs_bag2image**) also has an Openni2 sample program that has OpenCV as a requirement.

http://unanancyowen.com/en/rs2driver/
https://github.com/UnaNancyOwen/OpenNI2Sample

Novruz97, 2021-02-18:
Hello sir @MartyG-RealSense ,
Thank you for your reply!

I have checked the samples from your link. Although they are in C++, which I don't much understand. My labmate helped me to try to run the codes. 

I have read and compiled the Cmake in the "Wrappers/openni2" and I could get the driver for Openni2 tools to be able to find the device successfully. I could run some viewers of openni. 

However, as we ran the codes from @UnaNancyOwen samples, the `openni::ANY_DEVICE`, couldn't find my camera. So we thought to add the generated driver file into the system include/OpenNi2/Drivers. But the code didn't work either. 

So, I will try to use librealsense2 library with Opencv learning. 

If you have any idea what could go wrong, please notice me! Thank you!

MartyG-RealSense, 2021-02-18:
The link below has some advice about getting openni2 to detect a RealSense camera.

https://github.com/IntelRealSense/librealsense/issues/2697

closed 2021-03-03 by MartyG-RealSense