realsenseai/librealsense #8384: Can we apply canny edge detection on depth image of realsense D435 camera

Mukeshgowda, 2021-02-18:
* I am working on task for detecting objects from depth images of d435 camera which mainly captures objects of shape resembling long vectical bars/lines, reactangular planes including human detections. I am checking if canny edge detection could help me in detecting these features to differentiate object.
       * Canny gradient calculation and thresholding for digital images is calculated based on the color intensity which is 
           represented in numerical values of pixels. 
        * How can I implement the gradient thresholding for depth image where I have only depth information for a pixel. Is it 
           possible to calculate gradient with only the depth information? 

Any inputs and assistence is appreciated.

MartyG-RealSense, 2021-02-18:
A good starting point in investigating your case will be to direct you to a case from November 2020 where a RealSense user was using a D435 with canny edge detection and provided detailed descriptions of the processes that they were using.  This should hopefully answer your initial questions.

https://github.com/IntelRealSense/librealsense/issues/7364#issuecomment-727635090

Mukeshgowda, 2021-02-22:
Hi @MartyG-RealSense Thanks for your response. I checked the previous case which you referred to (#7364). User has mentioned that he has used the Canny edge detection for depth image and this gives me answer that canny can be used for depth images. But he has not mentioned any detailed information on the method how he has evaluated the edges, whether he has used only depth distance available or if he has converted depth image to gray scale for applying canny algorithm.
Further details on what method is applied for gradient calculation could be more helpful.
Thanks in advance for your answers.

MartyG-RealSense, 2021-02-22:
Hi @Mukeshgowda    I researched your question further.  There is very little information available about using RealSense 400 Series cameras with canny edge detection for depth image processing though.

An information source my research found that seemed to cover multiple aspects of what you are seeking (RealSense, edges, gradient, grayscale, etc) is linked to below. 

https://www.ncbi.nlm.nih.gov/pmc/articles/PMC7696567/

MartyG-RealSense, 2021-03-01:
Hi @Mukeshgowda   Do you require further assistance with this case, please?   Thanks!

Mukeshgowda, 2021-03-02:
Hello @MartyG-RealSense Thanks for your inputs and assistance on the issue until now. I am trying to implement Canny with the information I have till now, which would take an other week or so. Could you please keep the case open until next week? 
I would come back with some results and update you.

Your assistance is appreciated.

MartyG-RealSense, 2021-03-02:
Yes, it is no problem to keep the case open for a further time period.  Please do update here when you have results.  Good luck!

MartyG-RealSense, 2021-03-14:
Hi @Mukeshgowda   Do you have an update about your case that you can provide, please?  Thanks!

Mukeshgowda, 2021-03-22:
Hello @MartyG-RealSense Thanks for keeping the case open until now and sorry for late response.
I did some research on applying canny for depth images and I figured out the possible ways could be to convert depth image to gray scale to have color intensity value to each pixel of depth image. Mapping of color intensity value for each depth pixel is quite tedious job as the information available from depth image for calculationg specific color diffrentiation is very less.
Once we have color intensity mapped for each pixels of depth image, we can apply canny as in normal.

The other approach which I though was to have customised version of canny algorithm. Conventional canny works on edge detection based on the change in color intensity value with adjacent pixels. Foe depth image we could consider edge detection and clasification based on depth distance where valid color pixels posses depth distance and dark pixels have no depth distance. I am still figuring out on how this can achieved and not sure how efficient this method could be.

As of now I have put an halt on canny implementation and continue further in later stage of my work. I appreciate your assistance here and thanks for your inputs. I don' need any further help on this case for now and you may close the case.
I would create a new case in future if i need any help.

MartyG-RealSense, 2021-03-22:
Thanks so much @Mukeshgowda for sharing your experience in detail for the benefit of other members of the RealSense community!  I will close this case now as suggested - please do feel free to create a new case at a future date.   Good luck!

closed 2021-03-22 by MartyG-RealSense

Mukeshgowda, 2021-05-10:
Hello @MartyG-RealSense  I have implemented Canny on the depth image. Kindly let me know if I can share some infornation on that here with the case closed.

MartyG-RealSense, 2021-05-10:
Hi @Mukeshgowda   Yes you are welcome to share information on a closed case.  Thanks very much!

Mukeshgowda, 2021-05-10:
Thankyou @MartyG-RealSense .

As i mentioned in above comment that to apply Canny algorithm, we need image to be in gray scale format where each pixel has some intensity value ranging from 0-255.
I have used the depth information available from depth image where each pixel has depth distance associated with it. Based on the range of field of vision of camera, pixel distance can vary from 0 to maximum range. I tried to scale the values of pixel distance from 0-255. I have considered gray scale image Matrix(type : cv Mat) of same dimension (height, width) as the depth image and assigned the corresponding indexes of gray scale image with pixel distance values of depth image.
For example if pixel_index -15 has depth ditance of 200 cm inn depth image, corresponding pixel index -15 of gray scale image is assigned the value 200. If any pixel has distance beyond 255 cm, those values are scaled to have a highest intensity value 255.
(It is necessary to have pixel distance to be an interger value to match with pixel intensity range of 0-255).

The gray scale image matrix we get will have pixel intensity value from 0-255. We can directly apply Canny algorithm or any other image processing algorithms from openCV on the image. It worked for me and hope this information helps for others(if needed).

@MartyG-RealSense I have solution to my case now.  I appreciate your assistance on the case and thanks for all the inputs.

MartyG-RealSense, 2021-05-10:
You are very welcome @Mukeshgowda - thanks for sharing your detailed solution with the RealSense community.  :)