realsenseai/librealsense #8385: L515 Firmware updates 1.5.4.1 breaks unity support

stents, 2021-02-18:
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.4.1 | 
| Operating System & Version |   Win 10  | 
| SDK Version                            |  2.42.0.2845    | 
| Language                            |  Unity       | 
| Segment			|  AR       | 

All Unity samples except PointCloud Processing Blocks fail to work with Firmware version 1.5.4.1 on the L515. IR and color streams are not found, resulting in "ExternalException: rs2_pipeline_start_with_config" errors etc.

Realsense Viewer works fine, it is just the Unity integration that appears to break.

Rolling back to firmware 1.5.0.0 fixes the issue and all samples appear to work as expected.

RealSenseSupport, 2021-03-01:
Hi @stents 

Do you have any log files that you can share?

closed 2021-03-19 by RealSenseSupport

maloel, 2021-04-07:
Hi Matt (@stents),

In our testing we managed to reproduce, but once we downgraded to an earlier FW version and upgraded again, the problem went away completely: we can no longer reproduce it.

I don't know if it's still relevant to you, but maybe you can try.

Thanks!

stents, 2021-04-08:
Ah thanks @maloel !
I will test it next time I have access to one and let you know if that works, however things were working fine with the lower firmware (1.5.0.0) since then.