realsenseai/librealsense #8386: Failed to setup frame size

erkarim85, 2021-02-18:
----------------------------------------------------------------------------------------------------

| Required Info                                 |                                                                          |
|---------------------------------|------------------------------------------- |
| Camera Model                               |  D435i                                                             | 
| Operating System & Version       |   Linux (Ubuntu 18.04 LTS)                            | 
| Platform                                         | NVIDIA Jetson Nano                                      |
| Language                                       |  python                                                           | 
| Segment			                  |  Robot                                                             | 

### Issue Description

I have a problem with the camera. I am trying to set the frame size to 640x480 but it is not working:

`import pyrealsense2 as rs`
`...`
`config = rs.config()`
`config.enable_stream(rs.stream.color,640,480,rs.format.bgr8,15)`

I am trying to use this config function, but I am getting frames of 1280x720 instead. 
Any idea how to fix that?

MartyG-RealSense, 2021-02-18:
Hi @erkarim85  I cannot find any problem with the stream definition in the code snippet that you posted.  Would it be possible to post the full script, please?  

Also, which librealsense version and camera firmware version are you using?

erkarim85, 2021-02-18:
the pyrealsense2 version is 2.31.0
I am not sure how to get the firmware version

Here is the complete code:

`import pyrealsense2 as rs
import numpy as np
import cv2
import datetime

fps = 15
time = 20
x = 640
y = 480

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, x, y, rs.format.bgr8, fps)

pipeline.start()

fourcc = cv2.VideoWriter_fourcc(*'XVID')
out = cv2.VideoWriter('OneMinVid.avi',fourcc,fps,(x,y))

print(datetime.datetime.now())
try:
    l = 0
    for i in range(fps*time):
        frame = pipeline.wait_for_frames()
        Cframe = frame.get_color_frame()
        if not Cframe:
            continue
        color_image = np.asanyarray(Cframe.get_data())
        tmp = color_image[..., ::-1]
        out.write(tmp)
        if l == 0:
            print(tmp.shape)
            print(color_image.shape)
            l = 1

        cv2.imshow('i',tmp)
        cv2.waitKey(1)

finally:
    pipeline.stop()
    print(datetime.datetime.now())
    out.release()
`

MartyG-RealSense, 2021-02-18:
The first thing that stands out to me is that your pipeline start instruction does not have a reference to config in the brackets, so the program would not know to apply your defined configuration to the pipeline and would use the default configuration for the camera model instead.  Please try changing the pipeline start instruction to the one below:

**pipeline.start(config)**

erkarim85, 2021-02-18:
Awesome, thank you! This fixed it. I used it like you said and the video was in the right dimensions. 

However, I tried to move forward to use depth and color, but I got an error, below is my code:

fps = 15
time = 10
x = 640
y = 480

#max0 = 0

#depth_array = np.zeros((y,x,fps*time))

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, x, y, rs.format.bgr8, fps)
config.enable_stream(rs.stream.depth, y, x, rs.format.z16, fps)

pipeline.start(config)

fourcc = cv2.VideoWriter_fourcc(*'XVID')

out = cv2.VideoWriter('color.avi',fourcc,fps,(x,y))
outd = cv2.VideoWriter('depth.avi',fourcc,fps,(x,y))

But I got an error in this line: pipeline.start(config)

MartyG-RealSense, 2021-02-18:
Your depth config line is using **y, x** instead of x, y like the color definition does.  You are therefore requesting a resolution of 480x640, which the pipeline cannot fulfill because RealSense 400 Series cameras do not support that resolution.

erkarim85, 2021-02-18:
I see. I just read it somewhere that the depth should be reversed. It is working fine now. 
Thank you very much for your support.

closed 2021-02-18 by erkarim85

MartyG-RealSense, 2021-02-18:
You are very welcome - I'm pleased I could help!