realsenseai/librealsense #8388: Multicam in Python

marcoagbarreto, 2021-02-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   {Win 10 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {3D scanning}                          | 

### Issue Description

Issue #1089, could have been a good start, no success.

I'm working on 3D scanning using 5 D435i cameras and python. I have been trouble dealing with the multicam setup in the code. I've read some of the posted issues and suggestions from other users. My first attempt was to create separate streams as suggested in [ #1735](https://github.com/IntelRealSense/librealsense/issues/1735) but that didn't go well and caused problems. 

Looking at the python examples, I checked into the box_dimensioner_multicam with no succes at all, I tried to run the file [realsense_device_manager.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py) alone for testing but the code gets stuck in line [#L196](https://github.com/IntelRealSense/librealsense/blob/bc0910f8ba3c33307ff247a29dd2b9e9ef1b269d/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L196) when poll_frames is called, needing to restar the kernel every time. I had a dificult time trying to understand what poll_frames function is doing. I believe this approach would be the best one for managing and streaming 5 cameras simultaneously, as I see the effort on the code.

However, right now I'm trying to follow up the same logic used in the [rs-multicam.cpp](https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam) and translate it to python language. So far, I'm able to get the serials of all cameras and I get to enable the devices, at the moment I'm using only 2 for testing porpuses. The problem I'm facing right now is the streaming as I'm streaming both cameras into a single stream or same cv2 window. I know that I need to assign a unique windows for each serial but I don't know how to do it. My main goal is to save depth and color images for later analysis.

Any help or suggestions on the best way for the implementation would be much appreciated. 

I attached the a picture of the code in jupyterlab, the raw code and the picture of the overlapping stream.

[image]

[image]

```python
# Libraries
import pyrealsense2 as rs
import numpy as np
import cv2 as cv

ctx = rs.context() # Create librealsense context for managing devices
serials = []
if (len(ctx.devices) > 0):
    for dev in ctx.devices:
        print ('Found device: ', \
                dev.get_info(rs.camera_info.name), ' ', \
                dev.get_info(rs.camera_info.serial_number))
        serials.append(dev.get_info(rs.camera_info.serial_number))
else:
    print("No Intel Device connected")
    
    
pipelines = []
for serial in serials:
    pipe = rs.pipeline(ctx)
    cfg = rs.config()
    cfg.enable_device(serial)
    cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
    pipe.start(cfg)
    pipelines.append(pipe)
    
while True:
    for pipe in pipelines:
        
        print('hello') # just for debugging 
#         frameset  = pipe.poll_for_frames()
        frameset  = pipe.wait_for_frames()
        color_frame = frameset.get_color_frame()
        if not color_frame:
                continue
        color_image = np.asanyarray(color_frame.get_data())
        cv.imshow('RealSense', color_image)
        cv.waitKey(1)
```

MartyG-RealSense, 2021-02-19:
Hi @marcoagbarreto  If it is possible for your project to be set up so that each camera has its own computing board instead of all cameras attached to the same computer, then the **EtherSense** Python ethernet networking project may be worth considering.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

The RealSense-compatible commercial 3D scanning software **RecFusion Pro** also has multicam support if you have the budget for such software and are not reliant on the solution being Python-based.

https://www.recfusion.net/index.php/en/features

If you prefer to have all cameras connected to the same computer and using Python, the blogger **Rahul Vishwakarma** provides a tutorial that may meet your needs.  It is a Windows-based pyrealsense2 example that uses the device manager from box_dimensioner_multicam to open a separate window for each camera.

https://rahulvishwakarma.wordpress.com/2019/08/17/realsense-435i-depth-rgb-multi-camera-setup-and-opencv-python-wrapper-intel-realsense-sdk-2-0-compiled-from-source-on-win10/

marcoagbarreto, 2021-02-19:
@MartyG-RealSense Thanks for the reply, forgot to mention the **EtherSense** example, definetly something to check out. Unfortunetly I'm limited to 1 PC at the moment, but still an option for the near future.

I liked the suggestion of the **RecFusion Pro** and the number of cameras they support. I'm working on reconstruction algorithms, so going this path will surely kill the fun. Still, helps me to see how other people's implementations.

I'll check **Rahul Vishwakarma** to get a feel on the code. Let me work on that and I'll get back to you to share my code implementation.

MartyG-RealSense, 2021-02-19:
Thanks very much for the update @marcoagbarreto and I look forward to your next report - good luck!

marcoagbarreto, 2021-02-19:
Nothing new here, only adapting some of the examples codes and following the **rs-multicam** logic for python.

Basically the next block of code finds all the connected cameras, enables them, does a color to depth alignment and displays individual windows for each camera. If key 's' is pressed pair of images are stored for both cameras.

I might stick with **Rahul Vishwakarma** implementation using the device manager and adjust it to my application, but the code below might help somebody getting started with multicam setup in python.

@MartyG-RealSense Thank you very much for the help and pointing me in the right direction. Any comments are welcome, otherwise I think we can close it. :)

```python
# Libraries
import pyrealsense2 as rs
import numpy as np
import cv2 as cv

def findDevices():
    ctx = rs.context() # Create librealsense context for managing devices
    serials = []
    if (len(ctx.devices) > 0):
        for dev in ctx.devices:
            print ('Found device: ', \
                    dev.get_info(rs.camera_info.name), ' ', \
                    dev.get_info(rs.camera_info.serial_number))
            serials.append(dev.get_info(rs.camera_info.serial_number))
    else:
        print("No Intel Device connected")
        
    return serials, ctx

def enableDevices(serials, ctx, resolution_width = 640,resolution_height = 480, frame_rate = 30):
    pipelines = []
    for serial in serials:
        pipe = rs.pipeline(ctx)
        cfg = rs.config()
        cfg.enable_device(serial)
        cfg.enable_stream(rs.stream.depth, resolution_width, resolution_height, rs.format.z16, frame_rate)
        cfg.enable_stream(rs.stream.color, resolution_width, resolution_height, rs.format.bgr8, frame_rate)
        pipe.start(cfg)
        pipelines.append([serial,pipe])
        
    return pipelines

def Visualize(pipelines):
    align_to = rs.stream.color
    align = rs.align(align_to)

    for (device,pipe) in pipelines:
        # Get frameset of color and depth
        frames = pipe.wait_for_frames()
        # frames.get_depth_frame() is a 640x360 depth image
        # Align the depth frame to color frame
        aligned_frames = align.process(frames)

        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()

        # Validate that both frames are valid
        if not aligned_depth_frame or not color_frame:
            continue

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Render images
        depth_colormap = cv.applyColorMap(cv.convertScaleAbs(depth_image, alpha=0.03), cv.COLORMAP_JET)
        images = np.hstack((color_image, depth_colormap))

        cv.imshow('RealSense' + device, images)
        key = cv.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv.destroyAllWindows()
            return True
            
        # Save images and depth maps from both cameras by pressing 's'
        if key==115:
            cv.imwrite( str(device) + '_aligned_depth.png', depth_image)
            cv.imwrite( str(device) + '_aligned_color.png', color_image)
            print('Save')
            
def pipelineStop(pipelines):
    for (device,pipe) in pipelines:
        # Stop streaming
        pipe.stop() 
        
# -------Main program--------
        
serials, ctx = findDevices()

# Define some constants 
resolution_width = 640 # pixels
resolution_height = 480 # pixels
frame_rate = 30  # fps

pipelines = enableDevices(serials, ctx, resolution_width, resolution_height, frame_rate)

try:
    while True:
        exit = Visualize(pipelines)
        if exit == True:
            print('Program closing...')
            break
finally:
    pipelineStop(pipelines)
```

MartyG-RealSense, 2021-02-20:
I have not got anything to add.  Thanks very much @marcoagbarreto for sharing the above code with the RealSense community.  Good luck with your next project step!

closed 2021-02-20 by marcoagbarreto

ivomarvan, 2021-05-19:
I wrote an example of connecting multiple Realsense cameras in Python:
https://github.com/ivomarvan/samples_and_experiments/tree/master/Multiple_realsense_cameras
Enjoy it if you find it useful.

MartyG-RealSense, 2021-05-19:
Thanks so much @ivomarvan for sharing your project with the RealSense community!   :)