realsenseai/librealsense #8391: RealDence and NCS 2

RarDay, 2021-02-19:
Hi everyone and @MartyG-RealSense !

Camera Model | D435i
-- | --
Intel Model | Neural Compute Stick 2
Operating System & Version | Windows 10
Language | Python 3.7

I recognize faces using the camera and the realsense library (python). 
How can I speed up the recognition process with using NCS2, do I understand correctly that I can throw the recognition process to this module?
I tried to do this through OpenVINO, but it seemed to me that if I use python, I can somehow make it easier by adding a couple of lines of code

MartyG-RealSense, 2021-02-19:
Hi @RarDay   Have tyou seen the **MobileNet-SSD-RealSense** example in the **Community Projects** section of the SDK examples page please?  This is a list of example projects contributed by RealSense community members.  The **MobileNet-SSD-RealSense** example makes use of NCS2, Python and **FaceDetection**.

https://github.com/PINTO0309/MobileNet-SSD-RealSense

RarDay, 2021-02-19:
I remembered what the problem was.
`net = cv2.dnn.readNet('lrmodel/MobileNetSSD/MobileNetSSD_deploy.xml', 'lrmodel/MobileNetSSD/MobileNetSSD_deploy.bin')`
`Traceback (most recent call last): Build OpenCV with Inference Engine to enable loading models from Model Optimizer`

MartyG-RealSense, 2021-02-19:
This particular example program was most recently validated with **OpenVINO 2019 R3** at the end of 2019.  Are you using the 2019 version or OpenVINO 2020 please?

Given that the readme of this example was last updated 16 months ago at the time of writing this, whilst the RealSense OpenVINO wrapper has been updated much more recently, I wonder if an incompatibility with this example might have developed during those 16 months.

RarDay, 2021-02-24:
I think i found problem, my python can not find opencv:
When i install OpenVINO, it set in windows path:
C:\Program Files (x86)\Intel\openvino_2019.3.334\opencv\bin
C:\Program Files (x86)\Intel\openvino_2019.3.334\python\python3.7

but in console Python 3.7.8:
>>> import cv2
ModuleNotFoundError: No module named 'cv2'

community.intel.com/t5/Intel-Distribution-of-OpenVINO/Problem-with-Windows-10-OpenCV-Python-and-NCS2/m-p/1176553#M17377

MartyG-RealSense, 2021-02-24:
I could not find a past RealSense OpenVINO case that involved the error **ModuleNotFoundError: No module named 'cv2'**.  I do not have any further advice that can I suggest unfortunately.

RarDay, 2021-02-25:
Hmm okay, ask questions on community.intel.com
But today i install openvino on my Nano Jetson, maybe it work

MartyG-RealSense, 2021-02-25:
Sorry I could not be of more help in this particular case.

Intel's forums have an OpenVINO section here:

https://community.intel.com/t5/Intel-Distribution-of-OpenVINO/bd-p/distribution-openvino-toolkit

closed 2021-03-01 by RarDay