realsenseai/librealsense #8392: Problem running an example code on Raspbian

ivankostadinov, 2021-02-19:
Realsense D435 
Firmware 05.12.10.00

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux
Raspberry pi 4 4GB - camera plugged via USB3. realsense-viewer works fine

SDK installed following these instructions:
https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

### Issue Description

I was trying to execute a simple example from the librealsense examples folder
/home/pi/librealsense/examples/hello-realsense
I opened the rs-hello-realsense.cpp and executed it with Geany.
This example code is supposed to read data from the camera and output the distance of the object in the center of the field of view.
I get the following error when executing:

/tmp/geany_run_script_L0FOY0.sh: 7: /tmp/geany_run_script_L0FOY0.sh: ./rs-hello-realsense: not found

I don't have much experience with Pi so I suspect I'm missing something very obvious. Probably opening the file with Geany directly is not the right way of doing it.

How do I run a basic example code for realsense camera?

MartyG-RealSense, 2021-02-19:
Hi @ivankostadinov   When The CMake build instruction **-DBUILD_EXAMPLES=true** is used (as it is in the Acrobotic guide) then the SDK examples and tools are built along with librealsense and can be launched.

If you modify the code of an example then you can update its build by going to the **Build** directory and running the **make** instruction again.  It only updates what has been changed instead of rebuilding the entire SDK.

https://github.com/IntelRealSense/librealsense/issues/2946#issuecomment-449290237

Could you check the **/usr/local/bin** folder to see if the built versions of the examples and tools are located there please?

ivankostadinov, 2021-02-19:
@MartyG-RealSense MartyG-RealSense 

Thank you for your reply.

I was able to do
 sudo rs-hello-realsense
and this started the example. The next thing I tried was modifying the program (just the cout) to see if the changes would be applied. I changed the cout and overwrote the file in the same folder /home/pi/librealsense/examples/hello-realsense

I ran the make instructions as follows:

pi@raspberrypi:~/librealsense/build $  make

but this took over an hour and after completion I ran sudo rs-hello-realsense and my changes were not applied (the cout was without my changes).

in  /usr/local/bin folder I do see the examples

My goal here is to eventually write my own code and execute it. I'm trying with baby steps by using the example code and modifying it but even this hasn't worked so far.

> Hi @ivankostadinov When the CMake build instruction **-DBUILD_EXAMPLES=true** is used (as it is in the Acrobotic guide) then the SDK examples and tools are built along with librealsense and can be launched.
> 
> If you modify the code of an example then you can update its build by going to the **Build** directory and running the **make** instruction again. It only updates what has been changed instead of rebuilding the entire SDK.
> 
> [#2946 (comment)](https://github.com/IntelRealSense/librealsense/issues/2946#issuecomment-449290237)
> 
> You should be able to launch the examples as soon as librealsense has completed installation though without having to edit their code. Could you check the **/usr/local/bin** folder to see if the built versions of the examples and tools are located there please?

MartyG-RealSense, 2021-02-19:
I am not personally aware of an alternative answer about updating only the changes made to an example and not rebuilding the entire SDK.  

@ev-mp   Could you verify what the correct procedure is please for rebuilding only the changes made to a single example program in **Raspbian**, please?

In regard to creating your own application from the beginning, the process is somewhat different editing existing SDK examples.  A good starting point may be the **Getting Started** project, which is a self-contained project template.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

ivankostadinov, 2021-02-19:
I was able to build a fresh project with the steps below

mkdir build
cd build
cmake ../myProjectFolder
make -j4
./my-program-name

This build only that one project and not the whole SDK.

Thank you for your help!

> I am not personally aware of an alternative answer about updating only the changes made to an example and not rebuilding the entire SDK.
> 
> @ev-mp Could you verify what the correct procedure is please for rebuilding only the changes made to a single SDK example program in **Raspbian** instead of building the entire SDK again, please?
> 
> In regard to creating your own application from the beginning, the process is somewhat different to editing existing examples packaged with the librealsense SDK. A good starting point may be the **Getting Started** project, which is a self-contained project template.
> 
> https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

MartyG-RealSense, 2021-02-19:
You are very welcome.  It's great news that you were successful in updating just the edited project!

closed 2021-02-27 by MartyG-RealSense