realsenseai/librealsense #8395: Colorizer

oscarthorn, 2021-02-20:
Hi!

I've been trying to achieve the same effect as using the colorizer (with option 2 so it produces a grayscale image) but with opencv but have not quite been able to do it.

Do you happen to know how it can be done?

Thanks!

MartyG-RealSense, 2021-02-20:
Hi @oscarthorn   The OpenCV example program **rs-grabcuts** provides a demonstration of setting colourized Option 2.  

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/grabcuts

See lines  14-15 and 59-60 of **rs-grabcuts.cpp**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/grabcuts/rs-grabcuts.cpp

Further discussion about the use of the colorizer in rs-grabcuts can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/1223

oscarthorn, 2021-02-20:
Thanks for the reply but was I was after was tips for how to achieve the same effect as colorize but with only OpenCV. So without actually using colorizer the class.￼

MartyG-RealSense, 2021-02-20:
Someone took the approach of producing a grayscale depth image in an OpenCV-using project with cvtColor, converting to COLOR_RGB2GRAY:

** depth_gray_image = cv2.cvtColor(depth_color_image, cv2.COLOR_RGB2GRAY)**

https://github.com/IntelRealSense/librealsense/issues/5414

oscarthorn, 2021-02-20:
Yeah but that was after they already used colorizer, so it does not help me unfortunately. I want to use opencv to go from depth mat (with float values representing depth i meters) to a greyscale image (that looks like what the realsense viewer produces, that can be displayed.

MartyG-RealSense, 2021-02-20:
Could you provide more information about what you are trying to achieve, please?  It sounds as though you are trying not to use the librealsense SDK at all and instead access camera data directly through OpenCV.

oscarthorn, 2021-02-20:
Sure, I am using the sdk to get the depth data, then I'm transferring the depth data to another process/system. For various reasons I don't want to also transfer a colorized version of the depth data, so what I have is the cv::Mat containing the depth data.

MartyG-RealSense, 2021-02-20:
Below is another script that outputs gratscale without using colorizer:

https://www.programmersought.com/article/82984185981/

closed 2021-02-20 by oscarthorn