realsenseai/librealsense #8396: Installation error (missing file Python.h) on 2.42

zlite, 2021-02-20:
Installing 2.42 on a Jetson Nano, the make fails with this error (missing file):
```
/home/chris/librealsense-2.42.0/build/third-party/pybind11/include/pybind11/detail/common.h:122:10: fatal error: Python.h: No such file or directory
 #include <Python.h>
          ^~~~~~~~~~
compilation terminated.
```
The previous version of librealsense (2.41) works fine

MartyG-RealSense, 2021-02-20:
Hi @zlite  There were changes to pybind11 in 2.42.0.  Here are the details from the release notes:

**[Python] Integrate pybind11 V2.6.1 + replace pybind11 files with clone action
Clone pybind11 instead of keeping pybind files inside librealsense2 repo
Update pybind version 2.2.1 -> 2.6.1**

https://github.com/IntelRealSense/librealsense/pull/8176

May i confirm please what you are installing: is it the entire librealsense SDK, or only the Python wrapper?  And what installation process are you using?

zlite, 2021-02-20:
I'm following your instructions [here](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049).  This is for the entire librealsense SDK

MartyG-RealSense, 2021-02-20:
Thanks very much.  If the process works on 2.41.0 but not on 2.42.0 then I would recommend using 2.41.0 for the time being and see if others report the same problem in coming days.

Nir-Az, 2021-02-22:
Hi @zlite ,

please try to install python-dev package and try building again.
for example if you work with python 3.6:

sudo apt-get install python3.6-dev

closed 2021-03-07 by MartyG-RealSense