realsenseai/librealsense #8397: the 3D point clouds often jump...

timeanddoctor, 2021-02-20:
SR305 415 425

I tend to use pyrealsense2 to get the 3d coordinate of a skull model. But the three dimensional pointcloud alwasy change any time, even I used a maching holder. How to get a stable 3d image?

MartyG-RealSense, 2021-02-20:
Hi @timeanddoctor   Could you provide an RGB color image of the location where the camera is please so that I can see if there are lighting and environmental factors that could affect the depth measurement, please?  Thanks!

MartyG-RealSense, 2021-03-01:
Hi @timeanddoctor  Do you require further assistance with this case, please?  Thanks!

timeanddoctor, 2021-03-01:
Thank you! 
I will update the picture to show what I am doing.

MartyG-RealSense, 2021-03-01:
You are very welcome.  Please do provide an image.

MartyG-RealSense, 2021-03-07:
Hi @timeanddoctor  If you still require assistance with this case, can you provide an image please?  Thanks!

closed 2021-03-10 by timeanddoctor