realsenseai/librealsense #8398: Can I get a presice of 3d reconstruction under mm for intel depth camera?Which one ,I should buy?

timeanddoctor, 2021-02-20:
Can I get a presice of 3d reconstruction under mm for intel depth camera?Which one ,I should buy? I went to scan and print a model like pen.
Or ary there any tutorial to improve the precise. 
I bought the SR305 415 435。

MartyG-RealSense, 2021-02-20:
Hi @timeanddoctor   The link below has resources for commercial 3D scanning software that is compatible with the 400 Series cameras, including one that is compatible with SR305.

https://support.intelrealsense.com/hc/en-us/community/posts/360037943654-Photogrammetry-3D-scanning

For free open-source scanning options, you could create a **point cloud** scan (with the RealSense Viewer or a script of your own creation) that can be saved as a **ply** point cloud file.  That file can then be converted to a solid mesh model using free / open source software such as MeshLab or Pyntcloud.

http://www.meshlab.net/ 

https://github.com/daavoo/pyntcloud

MartyG-RealSense, 2021-03-01:
Hi @timeanddoctor  Do you require further assistance with this case, please?  Thanks!

timeanddoctor, 2021-03-01:
Thank you I bougth the SR305.

timeanddoctor, 2021-03-01:
bought 435 415 as well。

closed 2021-03-01 by timeanddoctor