realsenseai/librealsense #8400: Closer than min distance

oscarthorn, 2021-02-20:
Hi!

When an object gets closer to the camera than the minimum distance the depth map sometimes produces totally random values, often most of the object is set 0 for invalid distance but with lots of flickering points where the distance can be anything from 1 to 10 meters or even more.

Is it possible to filter or do something to make sure these close objects are just set to 0? In our application we cannot always control that objects do not get to close and these random values are a big problem.

MartyG-RealSense, 2021-02-20:
Hi @oscarthorn  You can set a **threshold filter** and define a minimum and maximum distance for the depth.  Values outside of that defined range will not be rendered on the depth image.  

https://github.com/IntelRealSense/librealsense/issues/5964#issuecomment-594372943

Alternatively, you can increase the value of the **Disparity Shift** option to reduce the minimum distance of the camera and enable the camera to get closer to objects before the image starts breaking up. Increasing the Disparity Shift value will also reduce the maximum observable depth distance though.

Using a D435 or D435i will enable objects to get closer to the camera by default, as these models have a minimum distance of 0.1 meters, compared to 0.3 meters for D415 and 0.4 meters for D455.

oscarthorn, 2021-02-22:
Hi!
The threshold filter does not really help since the noise is random so say the threshold filter is 0.5 to 10m, then the noise could be 6m which is equally bad as it being 30m. Like this:
[image]

Is this just something you have to live with?

MartyG-RealSense, 2021-02-22:
Hi @oscarthorn  I notice that you closed the case after your most reent question.  Did not mean to do this please?   If you meant to keep it open, please click the **Reopen Issue** button under the comment box.

oscarthorn, 2021-02-22:
No, sorry!

MartyG-RealSense, 2021-02-22:
You may be able to eliminate some of the bad data if you use the **High Accuracy** visual preset so that pixels with a low accuracy confidence are automatiucally excluded from the image.  This can make the image become more sparse in detail (more holes / gaps).  Increasing the **Laser Power** option's value above its default of '150' may help to compensate for this, as increasing Laser Power can make the depth image less sparse (whilst reducing Laser Power makes the depth image more sparse).

There will inevitably be bad data though if the camera is positioned right up against a surface though, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/7631#issuecomment-714504852

oscarthorn, 2021-02-22:
Thanks for the suggestions, I'll try them and see! Closing for real now :)

closed 2021-02-22 by oscarthorn