realsenseai/librealsense #8401: Librealsense with OpenCV Compatability

FinnMckinn, 2021-02-20:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I| 
| Firmware Version                   | 2.41.0.2657| 
| Operating System & Version |   {Win (10) |                                   | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.41.0 }                          | 
| Language                            |  {C++ }                          | 

### Issue Description
I am currently attempting to utilise OpenCV in my current program, however, I'm running into compatibility errors. Following /librealsense/tree/master/wrappers/opencv/grabcuts I will continually get an error stating "LNK1112: module machine type 'x64' conflicts with target machine type 'x86'". Now I understand the reasoning behind this error, being that OpenCV is x64 and librealsense is x86. though OpenCV has no x86 and librealsense has no x64, so I am incredibly confused on how I am meant to get the two to work together. 

If anyone could push me in the right direction I'd be very thankful, this has been causing me issues for a while.

MartyG-RealSense, 2021-02-20:
Hi @FinnMckinn  I studied your case carefully but I am not certain where the problem is occurring.  As a starting point in investigating your case, may I ask if you have installed OpenCV for librealsense using the Windows installation documentation for OpenCV in the link below?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/readme.md#windows

ev-mp, 2021-02-21:
@FinnMckinn hello,
>though OpenCV has no x86 and **librealsense has no x64**

Librealsense uses Cmake build system and can be configured to build 32 or 64-bit binaries on demand.
The selection of X64/X86 is done at CMake's config stage by passing appropriate flag to CMake generator:
[image]

Try to keep similar configuration settings for Librealsense and OpenCV (including release/debug flags) to mitigate linker and runtime inconsistencies.

MartyG-RealSense, 2021-03-05:
Hi @FinnMckinn   Do you require further assistance with this case, please?  Thanks!

FinnMckinn, 2021-03-05:
Using the method ev-mp explained I was able to get the libraries to work together. Thanks very much.

closed 2021-03-05 by FinnMckinn

MartyG-RealSense, 2021-03-05:
Great to hear @FinnMckinn  - thanks very much for the update!