realsenseai/librealsense #8402: how to use ue4 realsense plugin after instalation?

zuojun1991, 2021-02-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-02-21:
Hi @zuojun1991  May I ask which librealsense and Unreal Engine versions are you using, please?  If you are using librealsense **2.42.0** then that SDK version introduced UE wrapper changes that mean that it should be built against UE version **4.26**.  It introduces breaking changes for UE versions less than 4.26.

https://github.com/IntelRealSense/librealsense/pull/8036

MartyG-RealSense, 2021-03-01:
Hi @zuojun1991  Do you require further assistance with this case, please?  Thanks!

zuojun1991, 2021-03-04:
> Hi @zuojun1991 Do you require further assistance with this case, please? Thanks!

Thank you very much for your kindly reply. But I just give up and refund. Just bought iphone x. Iphone is IBM in 80s :(

zuojun1991, 2021-03-04:
@MartyG-RealSense

MartyG-RealSense, 2021-03-04:
Okay, thanks very much @zuojun1991  for the update.  I'm sorry that you were unable to achieve your goal.  Good luck with whatever you do next!

closed 2021-03-04 by MartyG-RealSense